l1npengtul/nokhwa #1: Example crashes on MacOS

teenjuna, 2021-05-31:
Hi! I tried to run the example on my MacBook, but encountered following error:
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: CouldntOpenDevice("Device at 0 not found")', src/main.rs:10:6
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

l1npengtul, 2021-05-31:
Please attach the code you were running along with the backtrace (RUST_BACKTRACE=1 cargo run).

l1npengtul, 2021-06-01:
Nevermind, I didn't read this properly. I do not have access to a mac with a webcam at this moment, I'll try when I get home.

l1npengtul, 2021-06-01:
I think this is a problem with `libuvc` not recognizing the integrated mac webcam. This should be fixed by the planned `ffmpeg` backend.

teenjuna, 2021-06-01:
All right then! Do you have a timeline for the `ffmpeg`? Asking out of curiosity :)

l1npengtul, 2021-06-01:
Next couple weeks. Right now its finals week, but I should be free after!

teenjuna, 2021-06-01:
Oh, I get it. Same for me 🌝

l1npengtul, 2021-07-02:
I just released 0.3.0 on crates, can you try that using the gstreamer backend? Maybe try and use sudo, since MacOS may be interfering with camera acquisiton.

l1npengtul, 2021-07-15:
Since OP has not replied, I will close tgis for now. If they respond I will reopen this issue.

teenjuna, 2021-07-17:
Hi, sorry for the long respond. I'm having troubles with the compilation of the provided `capture` example:
```
   Compiling gfx-backend-empty v0.9.0
   Compiling cocoa v0.24.0
error: failed to run custom build command for `v4l2-sys-mit v0.2.0`

Caused by:
  process didn't exit successfully: `/Users/virginiacool/.cargo/target/debug/build/v4l2-sys-mit-7c077684d105fde2/build-script-build` (exit code: 101)
  --- stderr
  wrapper.h:1:10: fatal error: 'linux/videodev2.h' file not found
  wrapper.h:1:10: fatal error: 'linux/videodev2.h' file not found, err: true
  thread 'main' panicked at 'Failed to generate bindings: ()', /Users/virginiacool/.cargo/registry/src/github.com-1ecc6299db9ec823/v4l2-sys-mit-0.2.0/build.rs:10:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
```

l1npengtul, 2021-07-17:
Sorry, forgot to add that you need to edit the `Cargo.toml` of the example.
Try changing the 
```
[dependencies.nokhwa]
path = "../../../nokhwa"
features = ["docs-only"]
```
to 
```
[dependencies.nokhwa]
path = "../../../nokhwa"
features = ["input-gst"]
```

l1npengtul, 2021-07-17:
This also assumes that you have gstreamer installed on your Mac.

teenjuna, 2021-07-17:
With installed gstreamer:
```
 Compiling gstreamer-app-sys v0.17.0
   Compiling gstreamer-base v0.17.0
The following warnings were emitted during compilation:

warning: `"pkg-config" "--libs" "--cflags" "gstreamer-app-1.0" "gstreamer-app-1.0 >= 1.8"` did not exit successfully: exit code: 1

error: failed to run custom build command for `gstreamer-app-sys v0.17.0`

Caused by:
  process didn't exit successfully: `/Users/virginiacool/.cargo/target/debug/build/gstreamer-app-sys-a9e1d757b06091c7/build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=GSTREAMER_APP_1.0_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=GSTREAMER_APP_1.0_STATIC
  cargo:rerun-if-env-changed=GSTREAMER_APP_1.0_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_aarch64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_aarch64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_aarch64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_aarch64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_aarch64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_aarch64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  cargo:warning=`"pkg-config" "--libs" "--cflags" "gstreamer-app-1.0" "gstreamer-app-1.0 >= 1.8"` did not exit successfully: exit code: 1
  --- stderr
  Package gstreamer-app-1.0 was not found in the pkg-config search path.
  Perhaps you should add the directory containing `gstreamer-app-1.0.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'gstreamer-app-1.0' found
  Package gstreamer-app-1.0 was not found in the pkg-config search path.
  Perhaps you should add the directory containing `gstreamer-app-1.0.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'gstreamer-app-1.0' found

warning: build failed, waiting for other jobs to finish...
The following warnings were emitted during compilation:

warning: `"pkg-config" "--libs" "--cflags" "gstreamer-video-1.0" "gstreamer-video-1.0 >= 1.8"` did not exit successfully: exit code: 1

error: build failed
```

l1npengtul, 2021-07-17:
Can you paste the output of 
`/Library/Frameworks/GStreamer.framework/Commands/gst-launch-1.0 --version`?

teenjuna, 2021-07-17:
I have it in another location: `/opt/homebrew/bin/gst-launch-1.0`, the version is:
```
gst-launch-1.0 version 1.18.4
GStreamer 1.18.4
Unknown package origin
```
I installed it with `brew install gstreamer`, so the location is the default one. It may be related to the fact that it's a M1 Mac, I heard they use `/opt/homebrew` for the ARM binaries

l1npengtul, 2021-07-17:
Make sure you have all required components
```
brew install gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly gst-libav
```
Add `/opt/homebrew/bin/` to `PKG_CONFIG_PATH` environment variable

teenjuna, 2021-07-17:
For some reason:
```
❯ cargo run -- -q V4L -c 0 -s -w 1920 -h 1080 --framerate 30 --format MJPG -b V4L -d
   Compiling nokhwa v0.3.1 (/private/tmp/testnokhwa/nokhwa)
error[E0599]: no variant or associated item named `Ffmpeg` found for enum `utils::CaptureAPIBackend` in the current scope
   --> /private/tmp/testnokhwa/nokhwa/src/query.rs:63:28
    |
63  |         CaptureAPIBackend::Ffmpeg => {
    |                            ^^^^^^ variant or associated item not found in `utils::CaptureAPIBackend`
    |
   ::: /private/tmp/testnokhwa/nokhwa/src/utils.rs:279:1
    |
279 | pub enum CaptureAPIBackend {
    | -------------------------- variant or associated item `Ffmpeg` not found here

error[E0599]: no variant or associated item named `Ffmpeg` found for enum `utils::CaptureAPIBackend` in the current scope
   --> /private/tmp/testnokhwa/nokhwa/src/query.rs:22:58
    |
22  |                         query_devices(CaptureAPIBackend::Ffmpeg)
    |                                                          ^^^^^^ variant or associated item not found in `utils::CaptureAPIBackend`
    |
   ::: /private/tmp/testnokhwa/nokhwa/src/utils.rs:279:1
    |
279 | pub enum CaptureAPIBackend {
    | -------------------------- variant or associated item `Ffmpeg` not found here

error[E0599]: no variant or associated item named `Ffmpeg` found for enum `utils::CaptureAPIBackend` in the current scope
   --> /private/tmp/testnokhwa/nokhwa/src/query.rs:35:58
    |
35  |                         query_devices(CaptureAPIBackend::Ffmpeg)
    |                                                          ^^^^^^ variant or associated item not found in `utils::CaptureAPIBackend`
    |
   ::: /private/tmp/testnokhwa/nokhwa/src/utils.rs:279:1
    |
279 | pub enum CaptureAPIBackend {
    | -------------------------- variant or associated item `Ffmpeg` not found here

error[E0599]: no variant or associated item named `Ffmpeg` found for enum `utils::CaptureAPIBackend` in the current scope
   --> /private/tmp/testnokhwa/nokhwa/src/query.rs:48:58
    |
48  |                         query_devices(CaptureAPIBackend::Ffmpeg)
    |                                                          ^^^^^^ variant or associated item not found in `utils::CaptureAPIBackend`
    |
   ::: /private/tmp/testnokhwa/nokhwa/src/utils.rs:279:1
    |
279 | pub enum CaptureAPIBackend {
    | -------------------------- variant or associated item `Ffmpeg` not found here

error[E0599]: no variant or associated item named `Ffmpeg` found for enum `utils::CaptureAPIBackend` in the current scope
   --> /private/tmp/testnokhwa/nokhwa/src/query.rs:64:70
    |
64  |             Err(NokhwaError::UnsupportedOperation(CaptureAPIBackend::Ffmpeg))
    |                                                                      ^^^^^^ variant or associated item not found in `utils::CaptureAPIBackend`
    |
   ::: /private/tmp/testnokhwa/nokhwa/src/utils.rs:279:1
    |
279 | pub enum CaptureAPIBackend {
    | -------------------------- variant or associated item `Ffmpeg` not found here

error: aborting due to 5 previous errors

For more information about this error, try `rustc --explain E0599`.
error: could not compile `nokhwa`

To learn more, run the command again with --verbose.
```

l1npengtul, 2021-07-17:
Try running from the `0.3.2` branch (`senpai` is currently broken)
Also, you need to run
```
cargo run -- -q GST -c 0 -s  -b GST -d
```

teenjuna, 2021-07-17:
```
❯ cargo run -- -q GST -c 0 -s  -b GST -d
   Compiling capture v0.1.0 (/private/tmp/testnokhwa/nokhwa/examples/capture)
error[E0432]: unresolved import `nokhwa::NetworkCamera`
 --> src/main.rs:7:69
  |
7 | use nokhwa::{query_devices, Camera, CaptureAPIBackend, FrameFormat, NetworkCamera};
  |                                                                     ^^^^^^^^^^^^^ no `NetworkCamera` in the root

error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: could not compile `capture`

To learn more, run the command again with --verbose.
```

l1npengtul, 2021-07-17:
Fixed fcc7047, should work now.

mofeing, 2021-08-19:
I have tried running the example at branch `0.3.2` but all I got is the following error:
```
Device at index 0: Name: Cámara FaceTime HD (integrada), Description: Video/Source, Extra: , Index: 0
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: CouldntOpenDevice("Failed to open pipeline with args autovideosrc location=/dev/video0 ! image/jpeg,width=640,height=480,framerate=15/1 ! appsink name=appsink async=false sync=false: could not link autovideosrc0 to appsink, autovideosrc0 can't handle caps image/jpeg, width=(int)640, height=(int)480, framerate=(fraction)15/1")', src/main.rs:157:88
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Disconnected', src/main.rs:287:41
```

Backtrace:
```
stack backtrace:
   0: rust_begin_unwind
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:515:5
   1: core::panicking::panic_fmt
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/panicking.rs:92:14
   2: core::result::unwrap_failed
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/result.rs:1355:5
   3: core::result::Result<T,E>::unwrap
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/result.rs:1037:23
   4: capture::main::{{closure}}
             at ./src/main.rs:157:21
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Disconnected', src/main.rs:287:41
stack backtrace:
   0: rust_begin_unwind
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:515:5
   1: core::panicking::panic_fmt
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/panicking.rs:92:14
   2: core::result::unwrap_failed
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/result.rs:1355:5
   3: core::result::Result<T,E>::unwrap
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/result.rs:1037:23
   4: capture::main::{{closure}}
             at ./src/main.rs:287:29
   5: <winit::platform_impl::platform::app_state::EventLoopHandler<T> as winit::platform_impl::platform::app_state::EventHandler>::handle_nonuser_event::{{closure}}
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/platform_impl/macos/app_state.rs:100:13
   6: winit::platform_impl::platform::app_state::EventLoopHandler<T>::with_callback
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/platform_impl/macos/app_state.rs:78:13
   7: <winit::platform_impl::platform::app_state::EventLoopHandler<T> as winit::platform_impl::platform::app_state::EventHandler>::handle_nonuser_event
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/platform_impl/macos/app_state.rs:99:9
   8: winit::platform_impl::platform::app_state::Handler::handle_nonuser_event
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/platform_impl/macos/app_state.rs:205:21
   9: winit::platform_impl::platform::app_state::AppState::handle_redraw
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/platform_impl/macos/app_state.rs:352:9
  10: winit::platform_impl::platform::view::draw_rect
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/platform_impl/macos/view.rs:353:9
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: <unknown>
  16: <unknown>
  17: <unknown>
  18: <unknown>
  19: <unknown>
  20: <unknown>
  21: <unknown>
  22: <unknown>
  23: <unknown>
  24: <unknown>
  25: <unknown>
  26: <unknown>
  27: <() as objc::message::MessageArguments>::invoke
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/objc-0.2.7/src/message/mod.rs:128:17
  28: objc::message::platform::send_unverified
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/objc-0.2.7/src/message/apple/mod.rs:27:9
  29: objc::message::send_message
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/objc-0.2.7/src/message/mod.rs:178:5
  30: winit::platform_impl::platform::event_loop::EventLoop<T>::run_return
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/platform_impl/macos/event_loop.rs:177:22
  31: winit::platform_impl::platform::event_loop::EventLoop<T>::run
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/platform_impl/macos/event_loop.rs:144:9
  32: winit::event_loop::EventLoop<T>::run
             at /Users/mofeing/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.25.0/src/event_loop.rs:154:9
  33: capture::main
             at ./src/main.rs:285:13
  34: core::ops::function::FnOnce::call_once
             at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/ops/function.rs:227:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

l1npengtul, 2021-08-20:
I am working on an AVFoundation backend for macos to sidestep this issue altogether.

l1npengtul, 2021-08-20:
Tracking issue: #8

l1npengtul, 2021-09-24:
I have created an AVFoundation backend, you may want to test it out. (You need to run with the XCode project)
I'll close this for now - this seems to work.

closed 2021-09-24 by l1npengtul