l1npengtul/nokhwa #2: Example possibly misleading

casey, 2021-06-01:
I wasn't sure, but I think the example might be misleading. In this code:

```rust
loop {
    println!(
        "{:?}, {:?}",
        camera.get_frame().unwrap().width(),
        camera.get_frame().unwrap().height()
    );
}
```

Wouldn't it be getting the width of frame N, and the height of frame N + 1?

So instead it should be:

```rust
loop {
    let frame = camera.get_frame().unwrap();
    println!("{}, {}", frame.width(), frame.height());
}
```

(Also switched to use `Display` instead of `Debug`, since I don't think it's necessary here.)

l1npengtul, 2021-06-01:
Oops, didn't think of that. Thanks!

l1npengtul, 2021-06-01:
Fixed by commit 8a1d220. Thanks for pointing this out!

closed 2021-06-01 by l1npengtul

casey, 2021-06-01:
You bet! The library looks great, by the way ^_^