l1npengtul/nokhwa #3: Read order error in yuyv422_to_rgb888

Roughack, 2021-06-24:
Was experimenting with capturing in FrameFormat::YUYV and the picture I got was green and pink.
The conversion function `yuyv422_to_rgb888` seems to read the color components in the wrong order.
It currently reads it as UYVY which mixes up the Chroma and Luma planes.

Here is a patch that solves it, with it my captured picture has the expected colors.
```diff --git a/src/utils.rs b/src/utils.rs
index 061bed8..c10c58e 100644
--- a/src/utils.rs
+++ b/src/utils.rs
@@ -340,7 +340,7 @@ pub fn yuyv422_to_rgb888(data: &[u8]) -> Result<Vec<u8>, NokhwaError> {
     let mut rgb_vec: Vec<u8> = vec![];
     if data.len() % 4 == 0 {
         for px_idx in (0..data.len()).step_by(4) {
-            let u = match data.get(px_idx) {
+            let y1 = match data.get(px_idx) {
                 Some(px) => match i32::try_from(*px) {
                     Ok(i) => i,
                     Err(why) => {
@@ -359,7 +359,7 @@ pub fn yuyv422_to_rgb888(data: &[u8]) -> Result<Vec<u8>, NokhwaError> {
                 }
             };
 
-            let y1 = match data.get(px_idx + 1) {
+            let u = match data.get(px_idx + 1) {
                 Some(px) => match i32::try_from(*px) {
                     Ok(i) => i,
                     Err(why) => {
@@ -378,7 +378,7 @@ pub fn yuyv422_to_rgb888(data: &[u8]) -> Result<Vec<u8>, NokhwaError> {
                 }
             };
 
-            let v = match data.get(px_idx + 2) {
+            let y2 = match data.get(px_idx + 2) {
                 Some(px) => match i32::try_from(*px) {
                     Ok(i) => i,
                     Err(why) => {
@@ -397,7 +397,7 @@ pub fn yuyv422_to_rgb888(data: &[u8]) -> Result<Vec<u8>, NokhwaError> {
                 }
             };
 
-            let y2 = match data.get(px_idx + 3) {
+            let v = match data.get(px_idx + 3) {
                 Some(px) => match i32::try_from(*px) {
                     Ok(i) => i,
                     Err(why) => {
```

l1npengtul, 2021-06-24:
Thanks for pointing that out, I will be pushing a patch soon!

l1npengtul, 2021-06-24:
Fixed by commit 38e99b6. Thank you for your contribution!

closed 2021-06-24 by l1npengtul