l1npengtul/nokhwa #6: Camera::frame_texture returns wgpu::TextureHandle instead of wgpu::Texture

mofeing, 2021-08-16:
I'm trying to use `nokhwa` in combination with `nannou` and I need to get a frame as a `wgpu::Texture`. As far I can see in the documentation, `Camera::frame_capture` returns a `WgpuTexture` which is an alias of `wgpu::Texture`.

In theory that would be alright. But when I call `frame_capture`, the compiler throws me the following error.

```
mismatched types
expected struct `nannou::wgpu::Texture`, found struct `TextureHandle`
```

As if the function is really returning a `wgpu::TextureHandle` object instead. 

I'm using:
- macOS 11.4 Big Sur
- `nannou` v0.17.1
- `nokhwa` v0.3.2
  - Features: `input-gst`, `output-wgpu`

l1npengtul, 2021-08-17:
The `nannou` crates's implementation creates a different `Texture` object per `nannou` documentation. As far as a pure Wgpu workload is concerned, it is working as intended.
Creating your own implementation of that method as a function should be trivial - just replace the `wgpu` to `nannou::wgpu`

closed 2021-08-17 by l1npengtul

mofeing, 2021-08-19:
So after digging up a lil bit (and some help from the `nannou` community), there is a better option than reimplementing the `Camera::frame_texture` method. Although this is a `nannou` thing, I'll post it here in case anyone runs into the same problem.

The idea is to capture a frame, transform the image into a `DynamicImage` and create a `nannou::Texture` using the `from_image` method.

```rust
let img = camera.frame().unwrap();
let dyn_img = image::DynamicImage::ImageRgb8(img);
let tex = nannou::wgpu::Texture::from_image((device, queue), &dyn_img);
```

l1npengtul, 2021-08-20:
I will add a `frame_nannou_texture` for `nannou` users.

mofeing, 2021-08-20:
Given that `nannou` is organized in a modular manner, I think it would be a better idea to write a wrapper and end implementing the features stated in nannou-org/nannou#464. Something like `nannou_camera` or `nannou_webcam`. 

I can work on this.