l1npengtul/nokhwa #12: Could not compile `nokhwa-bindings-windows`

thisjaiden, 2021-10-05:
I'm having this issue on both the latest stable and nightly branch.

Here's my dependencies section of my `Cargo.toml`:
`nokhwa = { version = "0.4.3", features = ["input-msmf", "decoding", "output-wgpu"] }`

And here's the error in more detail:
```
error[E0015]: calls in statics are limited to constant functions, tuple structs and tuple variants
   --> C:\Users\pizzajess\.cargo\registry\src\github.com-1ecc6299db9ec823\nokhwa-bindings-windows-0.2.4\src\lib.rs:372:47
    |
372 |     static mut INITIALIZED: Arc<AtomicBool> = Arc::new(AtomicBool::new(false));
    |                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0015`.
error: could not compile `nokhwa-bindings-windows` due to previous error
```

l1npengtul, 2021-10-06:
Will push version 0.5 to fix this.

closed 2021-10-06 by l1npengtul

l1npengtul, 2021-10-07:
Thanks! I look forward to the PR!