l1npengtul/nokhwa #15: Not able to access camera controls

oskargustafsson, 2021-11-03:
The docs show a struct called [CameraControls](https://docs.rs/nokhwa/0.8.0/nokhwa/struct.CameraControl.html), which contains a lot of useful functionality. Particularly, I'm interested in setting the exposure of a camera.

The problem is that I can't figure out how to access a camera's controls. Is it possible somehow?

l1npengtul, 2021-11-04:
The problem with CameraControls is that its implementation by each backend(MSMF, AVF, V4L2) is inconsistent, however you may have neglected [camera_control()](https://docs.rs/nokhwa/0.8.0/nokhwa/trait.CaptureBackendTrait.html#tymethod.camera_control) and its related methods. You have the best shot with the native backends for each OS, except AVFondation where I couldn't get through Apples documentation (PRs welcome!)

oskargustafsson, 2021-11-04:
Alright thanks, I didn't realize that backends could be explicitly instantiated. For posterity, the following code works for my purposes
```rust
let mut camera: Box<dyn CaptureBackendTrait + Sync + Send> = Box::new({
    #[cfg(unix)]
    { nokhwa::backends::capture::V4LCaptureDevice::new(0, None) }
    #[cfg(windows)]
    { nokhwa::backends::capture::MediaFoundationCaptureDevice::new(0, None) }
}?);
```

l1npengtul, 2021-11-04:
The `Camera` struct is a wrapper for the CaptureBackendTrait object. In the next release, I will add the necessary APIs to the `Camera` struct itself.

closed 2021-11-04 by l1npengtul