l1npengtul/nokhwa #17: Memory leakage occurs when too much delay per frame using `input-gst`

kerryeon, 2021-11-27:
Hi, I'm a fan of `nokhwa` library.

## Problem
* When using the `input-gst` feature, I found a problem that the memory usage increases indefinitely as the computation time per frame is longer than fps.

```rust
use nokhwa::{Camera, CameraFormat, FrameFormat};

fn main() {
    // set up the Camera
    let mut camera = Camera::new(
        0,
        Some(CameraFormat::new_from(640, 480, FrameFormat::MJPEG, 30)),
    )
    .unwrap();

    // open stream
    camera.open_stream().unwrap();
    loop {
        let frame = camera.frame().unwrap();
        println!("{}, {}", frame.width(), frame.height());

        // FIXME: Memory leakage occurs when too much delay per frame using `input-gst`
        std::thread::sleep(std::time::Duration::from_secs(9999));
    }
}

```

## Environment
* OS: Linux (ArchLinux)
* gstreamer: 1.18.5-1
* rustc: 1.56.1

## Expected reason
* There is no constraint on the frame buffer.

## Expected solution
* Set the size constraint of the frame buffer, and let the user control it.

l1npengtul, 2021-11-27:
The current issue is due to `GStreamerCaptureDevice` uses an `flume::unbounded()`. In 0.10 this will be updated to use an `Arc<Mutex<T>>` instead. Thank you for the report!

In the meantime, you can use the native OS bindings: `input-msmf`, `input-v4l`, `input-avfoundation`

l1npengtul, 2021-11-29:
#18 has been resolved, however it will be part of the 0.10 update so you will have to wait a while. 

I'll close this for now, feel free to make a new issue if the problem persists.

Thank you for the report!

closed 2021-11-29 by l1npengtul