l1npengtul/nokhwa #25: [Feature request] Support for greyscale captures

iamkroot, 2022-01-21:
Hey, thanks for creating this!
I'm trying to use this for capturing frames from the IR camera on my laptop (using v4l2), which gives 8-bit Greyscale stream.

Nokhwa errors out in `get_frame` since it expects an `rgb888` buffer. Using Opencv doesn't cause an error, but the captured image is still junk.

Would it be possible to handle (preferably automatically) different color formats (rgb24 vs grey8)?

l1npengtul, 2022-01-21:
This would be possible - we could add the color format to `CameraFormat` 
I will release this as 0.10 and push the current main branch to 0.11

l1npengtul, 2022-01-21:
The proposed API changes would be:
New `PixelBuffer` Enum Type added for `CameraFormat`
Custom buffer type for pixelbuffer built on top of `Cow[u8]`

l1npengtul, 2022-01-21:
It would be helpful if you listed the output of `v4l2-ctl --list-formats-ext` and `v4l2-ctl --all` of your IR camera.

iamkroot, 2022-01-22:
Here you go-

```
$ v4l2-ctl -d 2 -V --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)

```

```
$ v4l2-ctl -d 2 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : Video Capture 12
	Bus info         : usb-0000:04:00.3-3
	Driver version   : 5.15.12
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : USB2.0 HD UVC WebCam: USB2.0 IR
	Serial           : 0x0001
	Bus info         : usb-0000:04:00.3-3
	Media version    : 5.15.12
	Hardware revision: 0x00001964 (6500)
	Driver version   : 5.15.12
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : Video Capture 12
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000013: from remote pad 0x100000a of entity 'IR Microsoft Extended Controls ' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 9: ok)
Format Video Capture:
	Width/Height      : 640/360
	Pixel Format      : 'GREY' (8-bit Greyscale)
	Field             : None
	Bytes per Line    : 640
	Size Image        : 230400
	Colorspace        : sRGB
	Transfer Function : Rec. 709
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 360
	Default     : Left 0, Top 0, Width 640, Height 360
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 360, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 360, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
```

l1npengtul, 2022-05-22:
Fixed cea3de73a2d74857e14f8fb5750b86c1c860bd55. Please wait for 0.10.

closed 2022-05-22 by l1npengtul