l1npengtul/nokhwa #31: GetPropertyError when opening some cameras on Windows (MSMF backend)

hamaluik, 2022-03-08:
I get this error when attempting to open some cameras using the MSMF backend and I can't quite figure out what's going on.

Minimal code here:

```toml
[package]
name = "camdebug"
version = "1.0.0"
edition = "2021"

[dependencies]
nokhwa = { version = "0.9", features = ["input-msmf"] }
```

```rust
use nokhwa::*;

fn main() {
    let cameras = query().expect("can query cameras");
    println!("cameras: {:#?}", cameras);
    let camera = Camera::new(0, None);
    if let Err(e) = camera {
        eprintln!("camera open error: {:?}", e);
    }
}
```

This outputs the following:

```
    Finished dev [unoptimized + debuginfo] target(s) in 0.06s
     Running `target\debug\camdebug.exe`
cameras: [
    CameraInfo {
        human_name: "Iriun Webcam",
        description: "Media Foundation Device",
        misc: "\\\\?\\root#camera#0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\\{2a1f7d28-c666-4848-a04c-3b1a9f2546a2}",
        index: 0,
    },
    CameraInfo {
        human_name: "Integrated Webcam",
        description: "Media Foundation Device",
        misc: "\\\\?\\usb#vid_0c45&pid_672e&mi_00#6&18c3862&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\\global",
        index: 1,
    },
]
camera open error: GetPropertyError { property: "Devices", error: "Not Found" }
```

Any ideas what is going on and/or how to solve it? Thanks!

hamaluik, 2022-03-08:
Hmm.. I have been able to get it working by manually supplying a known good camera format and then after opening the camera, changing the format appropriately based on what formats are compatible, a la:

```rust
use nokhwa::*;

fn main() {
    let cameras = query().expect("can query cameras");
    println!("cameras: {:#?}", cameras);
    let camera = Camera::new(
        0,
        Some(CameraFormat::new(
            Resolution::new(640, 480),
            FrameFormat::YUYV,
            30,
        )),
    );
    match camera {
        Ok(mut c) => {
            let formats = c.compatible_camera_formats();
            println!("compatible formats: {:?}", formats);

            if let Ok(formats) = formats {
                let mut formats: Vec<CameraFormat> = formats
                    .into_iter()
                    .filter(|f| {
                        f.frame_rate() > 20
                            && f.frame_rate() < 40
                            && f.format() == FrameFormat::YUYV
                            && f.width() <= 1920
                    })
                    .collect();
                formats.sort_by(|a, b| a.width().cmp(&b.width()));
                let format = formats.pop().expect("can have a large format");
                println!("choosing format: {:#?}", format);
                c.set_camera_format(format).expect("can set frame format");

                println!("capturing frame");
                match c.frame() {
                    Ok(f) => println!("frame captured! {:?}", f.dimensions()),
                    Err(e) => println!("failed to capture frame: {:?}", e),
                }
            }
        }
        Err(e) => eprintln!("camera open error: {:?}", e),
    }
}
```

(which works fine and captures the frame successfully)

So I suspect something is going wrong with the "automatic format detection" when calling `Camera::new(index, None)`, or the API seems a bit unintuitive that I have to supply a camera format before I know what camera formats are supported?

l1npengtul, 2022-03-09:
Yes, the camera API sets the default format to 640x480, MJPEG at 15FPS (for a consistent experience). Considering changing it so that the open camera returns the camera format in 0.10

If you can do any more diggin, that would be great!

harry1453, 2022-05-26:
Hey!

I've just found that this only happens on cameras that do not have any MJPEG modes (only YUYV modes)

l1npengtul, 2022-05-26:
This is being fixed in 0.10

closed 2022-10-12 by l1npengtul