l1npengtul/nokhwa #34: Poor performance with v4l backend

SludgePhD, 2022-05-01:
I'm trying to port a library I wrote from my custom V4L2 bindings (https://github.com/SludgePhD/LinuxVideo) to nokhwa and am seeing reduced performance:

```
[2022-05-01T12:53:43Z DEBUG zaru::webcam] trying #3 'Logitech BRIO' (Video4Linux Device @ /dev/video3)
[2022-05-01T12:53:43Z INFO  zaru::webcam] started stream at 1280x720@90Hz
[2022-05-01T12:53:44Z DEBUG zaru::timer] webcam: 3 FPS (dequeue: 3x333.5ms, decode: 3x6.8ms)
[2022-05-01T12:53:45Z DEBUG zaru::timer] webcam: 29 FPS (dequeue: 29x25.3ms, decode: 29x9.8ms)
[2022-05-01T12:53:46Z DEBUG zaru::timer] webcam: 29 FPS (dequeue: 29x25.1ms, decode: 29x9.8ms)
[2022-05-01T12:53:47Z DEBUG zaru::timer] webcam: 29 FPS (dequeue: 29x22.9ms, decode: 29x11.7ms)
[2022-05-01T12:53:48Z DEBUG zaru::timer] webcam: 29 FPS (dequeue: 29x28.3ms, decode: 29x6.6ms)
[2022-05-01T12:53:49Z DEBUG zaru::timer] webcam: 29 FPS (dequeue: 29x28.5ms, decode: 29x6.6ms)
[2022-05-01T12:53:50Z DEBUG zaru::timer] webcam: 29 FPS (dequeue: 29x28.1ms, decode: 29x6.8ms)
```

`dequeue` is the average time taken by `camera.frame_raw()`, `decode` is the time mozjpeg needs to decode the JPEG frame (which I do by hand, hence `frame_raw()`). The result of this is correct, and the `decode` times remained about the same, but the performance of `dequeue` is a lot worse than before (and much lower than what the webcam can deliver or what was requested).

For comparison, with LinuxVideo I have no problems running at 86 FPS, which I think is as fast as the webcam outputs frames when requesting 90 Hz. Note that LinuxVideo sidesteps libv4l and directly calls the device ioctls. The `dequeue` time is essentially completely spent blocking on the device read.

It could be that libv4l does some expensive conversion, or that something with setting the frame rate is going wrong (though `camera.frame_rate()` does correctly return 90).

This is with nokhwa 0.9.4, not latest git, but I don't see much that could affect performance.

(side note, I have tried the gstreamer backend instead of `input-v4l`, but it fails with a nondescript `UnsupportedOperationError(Auto)`)

l1npengtul, 2022-05-01:
Are you using the camera in a dark or bright environment?

SludgePhD, 2022-05-01:
Pretty bright, it's not due to auto-exposure (it performs better with LinuxVideo in the same environment)

l1npengtul, 2022-05-02:
Im working on a fix in a local 0.10 branch.

D1plo1d, 2022-10-31:
@l1npengtul Did you have any success improving the v4l2 performance? I'm also seeing between 8 and 18 fps on a system that normally performs much faster.

l1npengtul, 2022-10-31:
> @l1npengtul Did you have any success improving the v4l2 performance? I'm also seeing between 8 and 18 fps on a system that normally performs much faster.
> 
> Edit: I am also seeing a low fps when I use the `v4l` crate directly. Maybe a problem on their end?
> 
> Edit 2: Nm, that was comparing MJPEG nokhwa vs a much slower v4l mode, changing v4l to MJPEG resulting in 20fps so there is something slow happening in nohawk specifically.

Maybe its frame decoding taking too long?

D1plo1d, 2022-11-01:
> Instead of frame, try frame_raw to see if you get the same performance

Yup, that was it. `frame_raw` gets the same fps as using v4l directly.

l1npengtul, 2022-11-01:
@D1plo1d You should be able to fix it by using Callback Camera. That runs the camera in another thread, allowing you to decode frames in a different thread from the camera.

l1npengtul, 2022-11-01:
This issue is a non-nokhwa issue. Closing.

closed 2022-11-01 by l1npengtul