l1npengtul/nokhwa #37: How to enumerate compatible formats of a camera index

aleokdev, 2022-05-21:
Hello! I have a webcam of unknown parameters and I'd like to have a list of its supported formats and resolutions. How can you achieve that with the crate? I see there's [`compatible_camera_formats`](https://docs.rs/nokhwa/latest/nokhwa/struct.Camera.html#method.compatible_camera_formats), but to call it you need a `Camera` already constructed. I can't construct it because I don't know the format I need to provide. What should I do? (BTW: I'm only using the v4l backend)

aleokdev, 2022-05-21:
Constructing a camera with `None` doesn't always work, as the default parameters don't seem to work with *very* webcam model.

l1npengtul, 2022-05-21:
This is being worked on in 0.10.

closed 2022-09-18 by l1npengtul