l1npengtul/nokhwa #46: Does not build on Apple Silicon

nikitavbv, 2022-06-21:
Hi, thank you for maintaining this crate!

It seems there is `symbol(s) not found for architecture arm64` when building on a MacBook with an M1 processor.

Here is the code I use:
```
use {
    std::{thread::sleep, time::Duration},
    nokhwa::{Camera, nokhwa_initialize, query},
};

fn main() {
    nokhwa_initialize(|result| {
        println!("on_complete: {}, query: {:?}", result, query()); 
        let camera = Camera::new(1, None).unwrap();
        println!("camera: {:?}", camera.info());
    });
    sleep(Duration::from_secs(10));
}
```

Cargo.toml:
```
[package]
name = "nokhwa-testing"
version = "0.1.0"
edition = "2021"

[dependencies]
nokhwa = { version = "0.9.4", features = ["input-avfoundation"] }
```

Here is a linker error I encounter:
```
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "-arch" "arm64" "/var/folders/k8/tc8kf6997r9gwl156j5b0g1h0000gn/T/rustcad2hl8/symbols.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.11b26thxr2zkmh75.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.14ju9lp2uv7xhil7.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.1isbi16a1tjg7o5f.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.3157rqi1un27yfph.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.339giqviijkbdou6.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.3kwn36whf96wn4nw.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.3oypbmz1qu6dv53c.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.3p3ucd0uz6i9kmmt.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.3q1aqfnimpcwrq6k.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.3qmqqgs3zmfmeqas.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.3wb3sk0nl9iivagx.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.417cdvo4hhcqac1g.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.49s8netqzp6lphje.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.4zqfqkynbefkt7uy.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.51smx5yvu2pz9xip.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.54eb0f5x3en2jk52.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.8wxashage992bxm.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.nfdq08phd3fnq4o.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.x7vl8kune1u8ikw.rcgu.o" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/nokhwa_testing-ee6d6ca36fb7e9d8.3yj7iby9c11nidld.rcgu.o" "-L" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps" "-L" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/build/mozjpeg-sys-f582892835282850/out" "-L" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/build/mozjpeg-sys-f582892835282850/out" "-L" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/build/objc_exception-c7958e843b8e62bd/out" "-L" "/Users/nikitavbv/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libnokhwa-1a079bdbbb5589a9.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libmozjpeg-6634d948351acdd4.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/librgb-480af82697f3e438.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libfallible_collections-3309b696366dad20.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libhashbrown-1d8e74aeb478f341.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libahash-5f759626124e5b0b.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libonce_cell-4e22a751c4ad0e53.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libarrayvec-efc691a55c0983c5.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libmozjpeg_sys-09fe7e07ea2f7db4.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libnokhwa_bindings_macos-4cd999170c2a10bf.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libflume-ed4efa3fd1d7557e.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libspin-2407881c60aa70b0.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/liblock_api-7fbb72b6cfc1b0c6.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libscopeguard-eb1e61010332a1cf.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libpin_project-a98907d1fcb05856.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libfutures_sink-40b09e67990431db.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libfutures_core-1b3a27dc641a732a.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libnanorand-a27a2d340e69126f.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libgetrandom-732ae10a5aefa5d2.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libdashmap-a83262b3454e3e27.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libnum_cpus-3b913b51fe59fbf0.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcfg_if-e6509c5f025d8702.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcocoa_foundation-f4d5900a80da637b.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libforeign_types-b1a73fdbd94e948a.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libforeign_types_shared-b2b813634eacbc55.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcore_graphics_types-2e141c6049806b3c.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcore_foundation-8928320c492b4635.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcore_foundation_sys-a98884409e1a2efe.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libbitflags-295066ef4f1a08fb.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libblock-54d07849328ebd85.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcore_media_sys-06f8f5185bcc4631.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcore_foundation_sys-d16b009260a549e9.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcfg_if-504e5b6598191998.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libthiserror-0b5d3046e86037a8.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/liblazy_static-f75dab69ff4510ec.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libobjc-0959b0f3a1d78466.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libobjc_exception-08e7865472ccbbe2.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libmalloc_buf-36827c8d724d6279.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/liblibc-1b5ce4b92aa38abf.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libimage-79c28f91dd58bf9a.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libcolor_quant-22192260062230e7.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libnum_iter-7d6461ad6c5f1c6d.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libbytemuck-88e115a31e194307.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libnum_rational-84c9edafed28713c.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libnum_integer-824e1dd12fc03d2b.rlib" "/Users/nikitavbv/dev/nokhwa-testing/target/debug/deps/libnum_traits-4b4376f67277d6cf.rlib" "/Users/nikitavbv/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libstd-af3f3292c66eb7cc.rlib" "/Users/nikitavbv/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libpanic_unwind-bc47a5a819240eac.rlib" "/Users/nikitavbv/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libobject-5fa6d29c50391a26.rlib" "/Users/nikitavbv/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libmemchr-9f34659cc49c55cf.rlib" "/Users/nikitavbv/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/ [... cut ...]
                _$LT$nokhwa_bindings_macos..avfoundation..CALLBACK_CLASS$u20$as$u20$core..ops..deref..Deref$GT$::deref::__static_ref_initialize::capture_out_callback::hbcd689a01a46737c in libnokhwa_bindings_macos-4cd999170c2a10bf.rlib(nokhwa_bindings_macos-4cd999170c2a10bf.nokhwa_bindings_macos.51d10249-cgu.1.rcgu.o)
          ld: symbol(s) not found for architecture arm64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)

error: could not compile `nokhwa-testing` due to previous error
```

Any ideas on a possible workaround? Thank you!

l1npengtul, 2022-06-21:
I am setting up a MacOSX vm, Ill see what I can do. If you know an easy way to get an M1 VM, let me know.

nikitavbv, 2022-06-21:
> If you know an easy way to get an M1 VM, let me know.

I think [hetzner offers M1 servers](https://www.hetzner.com/dedicated-rootserver/mac-mini-m1), as well as [AWS](https://aws.amazon.com/about-aws/whats-new/2021/12/amazon-ec2-m1-mac-instances-macos/) or [Scaleway](https://www.scaleway.com/en/hello-m1/).

Alternatively, If you have any ideas, you can post them here, and I will try it on my laptop. 
We can also debug this together during a video call if you prefer :)

l1npengtul, 2022-06-22:
Keep in mine that the 0.9.4 version is no longer being worked on, I am working on a fix for AVFoundation on 0.10

skahl, 2022-08-18:
Maybe this issue is related? https://github.com/rust-lang/rust/issues/86658#issuecomment-983925155

l1npengtul, 2022-09-28:
I cannot recreate this issue on latest commit. Closing.

closed 2022-09-28 by l1npengtul