l1npengtul/nokhwa #47: Can't get frame from camera

Meister1593, 2022-07-03:


Meister1593, 2022-07-03:
Sorry for empty issue at first, pressed enter too fast

l1npengtul, 2022-07-03:
I suspect this may be a your computer issue, Ill look into this

Meister1593, 2022-07-03:
> I suspect this may be a your computer issue, Ill look into this

I could open v4l camera through other library directly (/dev/video1) and with ffmpeg CLI, so I doubt that its issue with my system configuration

l1npengtul, 2022-07-04:
Can you attach the output of `v4l2-ctl --all`?

Meister1593, 2022-07-04:
This worked if i specified device directly
```
[plyshka@fedora-36 plyshka]$ v4l2-ctl -d /dev/video1 --all
Driver Info:
	Driver name      : ov534
	Card type        : USB Camera-B4.09.24.1
	Bus info         : usb-0000:01:00.0-9
	Driver version   : 5.18.8
	Capabilities     : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (ov534: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'GRBG' (8-bit Bayer GRGR/BGBG)
	Field             : None
	Bytes per Line    : 640
	Size Image        : 307200
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 2

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=0 value=0 flags=slider
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32 flags=slider
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=64 value=64 flags=slider
                            hue 0x00980903 (int)    : min=-90 max=90 step=1 default=0 value=0 flags=slider
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                       exposure 0x00980911 (int)    : min=0 max=255 step=1 default=120 value=120 flags=inactive, volatile
                 gain_automatic 0x00980912 (bool)   : default=1 value=1 flags=update
                           gain 0x00980913 (int)    : min=0 max=63 step=1 default=20 value=20 flags=inactive, volatile
                horizontal_flip 0x00980914 (bool)   : default=0 value=0
                  vertical_flip 0x00980915 (bool)   : default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=1 default=0 value=0 (Disabled)
				0: Disabled
				1: 50 Hz
                      sharpness 0x0098091b (int)    : min=0 max=63 step=1 default=0 value=0 flags=slider

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=1 default=0 value=0 (Auto Mode) flags=update
				0: Auto Mode
				1: Manual Mode
```

l1npengtul, 2022-07-04:
try `v4l2-ctl -d /dev/video1 --list-formats-ext`

Meister1593, 2022-07-04:
> `v4l2-ctl -d /dev/video1 --list-formats-ext`
```
[plyshka@fedora-36 plyshka]$ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GRBG' (8-bit Bayer GRGR/BGBG)
		Size: Discrete 320x240
			Interval: Discrete 0.005s (187.000 fps)
			Interval: Discrete 0.007s (150.000 fps)
			Interval: Discrete 0.007s (137.000 fps)
			Interval: Discrete 0.008s (125.000 fps)
			Interval: Discrete 0.010s (100.000 fps)
			Interval: Discrete 0.013s (75.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.020s (50.000 fps)
			Interval: Discrete 0.027s (37.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.020s (50.000 fps)
			Interval: Discrete 0.025s (40.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 320x240
			Interval: Discrete 0.005s (187.000 fps)
			Interval: Discrete 0.007s (150.000 fps)
			Interval: Discrete 0.007s (137.000 fps)
			Interval: Discrete 0.008s (125.000 fps)
			Interval: Discrete 0.010s (100.000 fps)
			Interval: Discrete 0.013s (75.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.020s (50.000 fps)
			Interval: Discrete 0.027s (37.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.020s (50.000 fps)
			Interval: Discrete 0.025s (40.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
```

Meister1593, 2022-07-04:
this is how i tried initializing it:
```
        let mut camera = Camera::with_backend(
            1,
            Some(CameraFormat::new_from(640, 480, FrameFormat::YUYV, 15)),
            CaptureAPIBackend::Video4Linux,
        )
```

Meister1593, 2022-07-04:
For reference: this is PlayStation Eye camera

l1npengtul, 2022-11-06:
I can't reproduce this, nor do I have an Eye camera...
Sorry, but I don't think I can fix this.

closed 2022-11-06 by l1npengtul