l1npengtul/nokhwa #48: unresolved import `windows::Win32::Foundation::PWSTR`

Aveline67, 2022-07-03:
```
   Compiling nokhwa-bindings-windows v0.3.4 (https://github.com/l1npengtul/nokhwa?branch=senpai#41abb00f)
error[E0432]: unresolved import `windows::Win32::Foundation::PWSTR`
   --> C:\Users\avlin\.cargo\git\checkouts\nokhwa-66b597f0abb1b795\41abb00\nokhwa-bindings-windows\src\lib.rs:382:13
    |
382 |             Foundation::PWSTR,
    |             ^^^^^^^^^^^^^^^^^ no `PWSTR` in `Windows::Win32::Foundation`
error: could not compile `nokhwa-bindings-windows` due to previous error
```

l1npengtul, 2022-07-03:
It is being worked on.

Aveline67, 2022-07-27:
How could i help ?

l1npengtul, 2022-07-27:
Im working on it, 0.10 should be ready for  release next week or so

closed 2022-09-18 by l1npengtul