l1npengtul/nokhwa #50: UnsupportedOperatorError on query_devices

ogxd, 2022-07-13:
Hello, I am having an `UnsupportedOperatorError` when I call `query_device`. The error has no internal message. I tried with the Auto backend and then on several other backends but I ended up having the same error everytime.

```rust
match nokhwa::query_devices(nokhwa::CaptureAPIBackend::Auto) {
    Ok(cameras) => {
        // Never gets here
    },
    Err(err) => {
        // Always an error
    }
}
```

[image]

- Version used: nokhwa 0.9.4
- OS: Windows 11 x64
- Dell XPS 15" from 2020

Any idea of what could go wrong?

l1npengtul, 2022-07-13:
Did you enable `input-msmf`?

ogxd, 2022-07-13:
No! Just saw this 🤦 
> The default feature includes nothing. Anything starting with input-* is a feature that enables the specific backend. As a general rule of thumb, you would want to keep at least input-uvc or other backend that has querying enabled so you can get device information from nokhwa.

It works as excepted now!

Maybe for others that don't take the time to read everything it could be interesting to put a message in the error mentionning that there are no backends?

l1npengtul, 2022-07-13:
Glad it worked out, I'll consider it for 0.10

closed 2022-07-13 by l1npengtul