l1npengtul/nokhwa #51: Capture example fails on invalid color format

bschwind, 2022-07-13:
OS: MacOS 12.3.1
Hardware: Apple Macbook Pro (14 inch, 2021), M1 Pro Chip
Rust Version: rustc 1.62.0 (a8314ef7d 2022-06-27)

Example invocation:

```
cargo run --release --features "input-avfoundation" -- --format YUYV
```

Output:

```
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: GetPropertyError { property: "875704438", error: "Invalid Value" }', src/main.rs:195:88
stack backtrace:
   0: rust_begin_unwind
             at /rustc/a8314ef7d0ec7b75c336af2c9857bfaf43002bfc/library/std/src/panicking.rs:584:5
   1: core::panicking::panic_fmt
             at /rustc/a8314ef7d0ec7b75c336af2c9857bfaf43002bfc/library/core/src/panicking.rs:142:14
   2: core::result::unwrap_failed
             at /rustc/a8314ef7d0ec7b75c336af2c9857bfaf43002bfc/library/core/src/result.rs:1785:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
Thread terminated, closing!
```

`875704438` corresponds to a FourCC of `420v`, or `CV420YpCbCr8BiPlanarVideoRange`

This error comes from this match block:

https://github.com/l1npengtul/nokhwa/blob/bd8e78134a123dc5fd3685264468ad27e99ddb17/nokhwa-bindings-macos/src/lib.rs#L862-L871

I suppose this is due to the newer facetime camera in the 2021 macbook pros?

If it's just a matter of adding support for an extra color format then I'd be happy to take a crack at it. Let me know if you need any help testing on apple hardware, I have a macbook pro 2014 (intel), macbook pro 2021 (M1 pro), and an M1 mac mini.

l1npengtul, 2022-07-13:
Ill keep this in mind since I am fixing MacOS code right now. It should be broken since 0.10 is still a WIP. If you want a working version use 0.9.2 branch (actually 0,9.4)

bschwind, 2022-07-13:
Ah sorry! I forgot to mention I'm on commit 5a8d9c3 . I'll edit my original post.

l1npengtul, 2022-11-05:
This has been fixed in the lastest `senpai` by adding NV12 support for MacOS

closed 2022-11-05 by l1npengtul

bschwind, 2022-11-05:
Thanks, @l1npengtul, I'll try to get this tested soon.

bschwind, 2022-11-07:
I realized I originally tested this on `examples/capture/src/main.rs`, but that is now an empty `main()` function.

I guess I can try re-adding the code, but I'm not sure if there's something catastrophically wrong with what was there before.

l1npengtul, 2022-11-07:
I'm rewriting the whole thing given the massive API changes of 0.10 :p

bschwind, 2022-11-07:
Makes sense! I'll get to testing it a bit later then once I put together a little program that uses it.

commit 745d3f03 (2022-11-05): Add NV12 support for MacOS, fix #51, fix #52