l1npengtul/nokhwa #52: Error when running 0.10 example with AVFoundation

morajabi, 2022-07-15:
I execute this command: 
```
nokhwa/examples/capture on  senpai via 🦀 v1.61.0
❯ cargo run --features "input-avfoundation"
```

and it gives 14 errors: 
```
 cargo run --features "input-avfoundation"

   Compiling nokhwa v0.10.0 (/Users/mo/dev/nokhwa)
error[E0412]: cannot find type `Output` in module `crate::pixel_format`
   --> /Users/mo/dev/nokhwa/src/backends/capture/avfoundation.rs:275:50
    |
275 |     ) -> Result<ImageBuffer<crate::pixel_format::Output, Vec<u8>>, NokhwaError> {
    |                                                  ^^^^^^ not found in `crate::pixel_format`
    |
help: consider importing this struct
    |
17  | use std::process::Output;
    |

error[E0425]: cannot find value `cp` in this scope
   --> /Users/mo/dev/nokhwa/src/camera.rs:454:44
    |
454 |                           Ok(cap) => Some(Ok(cp.into())),
    |                                              ^^ help: a local variable with a similar name exists: `cap`
...
549 | / cap_impl_fn! {
550 | |     // (GStreamerCaptureDevice, new, feature = "input-gst", gst),
551 | |     (OpenCvCaptureDevice, new_autopref, feature = "input-opencv", opencv),
552 | |     // (UVCCaptureDevice, create, feature = "input-uvc", uvc),
...   |
555 | |     (AVFoundationCaptureDevice, new, all(feature = "input-avfoundation", any(target_os = "macos", target_os = "ios")), avfoundation)
556 | | }
    | |_- in this macro invocation
    |
    = note: this error originates in the macro `cap_impl_fn` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0412]: cannot find type `MSMF` in this scope
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:37:5
   |
37 |     MSMF,
   |     ^^^^ not found in this scope
   |
help: there is an enum variant `crate::BackendsEnum::MSMF`; try using the variant's enum
   |
37 |     crate::BackendsEnum,
   |     ~~~~~~~~~~~~~~~~~~~

error[E0412]: cannot find type `AVF` in this scope
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:38:5
   |
38 |     AVF,
   |     ^^^ not found in this scope
   |
help: there is an enum variant `crate::BackendsEnum::AVF`; try using the variant's enum
   |
38 |     crate::BackendsEnum,
   |     ~~~~~~~~~~~~~~~~~~~

error[E0412]: cannot find type `V4L2` in this scope
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:39:5
   |
39 |     V4L2,
   |     ^^^^ not found in this scope
   |
help: there is an enum variant `crate::BackendsEnum::V4L2`; try using the variant's enum
   |
39 |     crate::BackendsEnum,
   |     ~~~~~~~~~~~~~~~~~~~

error[E0412]: cannot find type `OCV` in this scope
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:40:5
   |
40 |     OCV,
   |     ^^^ not found in this scope
   |
help: there is an enum variant `crate::BackendsEnum::OCV`; try using the variant's enum
   |
40 |     crate::BackendsEnum,
   |     ~~~~~~~~~~~~~~~~~~~

error[E0412]: cannot find type `MSMF` in this scope
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:50:1
   |
50 | #[enum_dispatch(BackendsEnum)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope
   |
   = note: this error originates in the attribute macro `enum_dispatch` (in Nightly builds, run with -Z macro-backtrace for more info)
help: there is an enum variant `crate::BackendsEnum::MSMF`; try using the variant's enum
   |
50 | crate::BackendsEnum
   |

error[E0412]: cannot find type `AVF` in this scope
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:50:1
   |
50 | #[enum_dispatch(BackendsEnum)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope
   |
   = note: this error originates in the attribute macro `enum_dispatch` (in Nightly builds, run with -Z macro-backtrace for more info)
help: there is an enum variant `crate::BackendsEnum::AVF`; try using the variant's enum
   |
50 | crate::BackendsEnum
   |

error[E0412]: cannot find type `V4L2` in this scope
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:50:1
   |
50 | #[enum_dispatch(BackendsEnum)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope
   |
   = note: this error originates in the attribute macro `enum_dispatch` (in Nightly builds, run with -Z macro-backtrace for more info)
help: there is an enum variant `crate::BackendsEnum::V4L2`; try using the variant's enum
   |
50 | crate::BackendsEnum
   |

error[E0412]: cannot find type `OCV` in this scope
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:50:1
   |
50 | #[enum_dispatch(BackendsEnum)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope
   |
   = note: this error originates in the attribute macro `enum_dispatch` (in Nightly builds, run with -Z macro-backtrace for more info)
help: there is an enum variant `crate::BackendsEnum::OCV`; try using the variant's enum
   |
50 | crate::BackendsEnum
   |

error[E0412]: cannot find type `ControlValue` in this scope
   --> /Users/mo/dev/nokhwa/src/utils.rs:984:40
    |
984 |     pub fn set_value(&mut self, value: ControlValue) -> Result<(), NokhwaError> {
    |                                        ^^^^^^^^^^^^ not found in this scope

error[E0412]: cannot find type `ControlValue` in this scope
   --> /Users/mo/dev/nokhwa/src/utils.rs:994:36
    |
994 |     pub fn with_value(self, value: ControlValue) -> Result<Self, NokhwaError> {
    |                                    ^^^^^^^^^^^^ not found in this scope

error[E0574]: expected struct, variant or union type, found enum `ControlDescription`
   --> /Users/mo/dev/nokhwa/src/utils.rs:997:12
    |
997 |         Ok(ControlDescription {
    |            ^^^^^^^^^^^^^^^^^^ not a struct, variant or union type

warning: unused import: `nokhwa_initialize`
  --> /Users/mo/dev/nokhwa/src/backends/capture/avfoundation.rs:18:33
   |
18 |     mjpeg_to_rgb, nokhwa_check, nokhwa_initialize, yuyv422_to_rgb, CameraControl, CameraFormat,
   |                                 ^^^^^^^^^^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: unused import: `std::process::Output`
  --> /Users/mo/dev/nokhwa/src/backends/capture/avfoundation.rs:27:5
   |
27 | use std::process::Output;
   |     ^^^^^^^^^^^^^^^^^^^^

warning: unused imports: `any::Any`, `ops::Deref`
  --> /Users/mo/dev/nokhwa/src/backends/capture/avfoundation.rs:28:11
   |
28 | use std::{any::Any, borrow::Borrow, borrow::Cow, collections::HashMap, ops::Deref};
   |           ^^^^^^^^                                                     ^^^^^^^^^^

warning: unused import: `pixel_format::PixelFormat`
  --> /Users/mo/dev/nokhwa/src/camera.rs:18:21
   |
18 |     buffer::Buffer, pixel_format::PixelFormat, BackendsEnum, CameraControl, CameraFormat,
   |                     ^^^^^^^^^^^^^^^^^^^^^^^^^

warning: unused import: `frame_formats`
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:19:5
   |
19 |     frame_formats,
   |     ^^^^^^^^^^^^^

warning: unused imports: `RgbaImage`, `buffer::ConvertBuffer`
  --> /Users/mo/dev/nokhwa/src/camera_traits.rs:26:13
   |
26 | use image::{buffer::ConvertBuffer, ImageBuffer, RgbaImage};
   |             ^^^^^^^^^^^^^^^^^^^^^               ^^^^^^^^^

error[E0204]: the trait `Copy` may not be implemented for this type
   --> /Users/mo/dev/nokhwa/src/utils.rs:941:10
    |
941 | #[derive(Copy, Clone, Debug, Hash, Eq, PartialEq)]
    |          ^^^^
...
945 |     name: String,
    |     ------------ this field does not implement `Copy`
946 |     value: ControlDescription,
    |     ------------------------- this field does not implement `Copy`
947 |     flag: Vec<KnownCameraControlFlag>,
    |     --------------------------------- this field does not implement `Copy`
    |
    = note: this error originates in the derive macro `Copy` (in Nightly builds, run with -Z macro-backtrace for more info)

Some errors have detailed explanations: E0204, E0412, E0425, E0574.
For more information about an error, try `rustc --explain E0204`.
warning: `nokhwa` (lib) generated 6 warnings
error: could not compile `nokhwa` due to 14 previous errors; 6 warnings emitted
```

l1npengtul, 2022-07-15:
Please use the 0.9.4 release on crates.io availible on the 0.9.2 branch. 0.10 doesn't work yet.

l1npengtul, 2022-07-15:
Ah yes, that is due to the labyrinth that is Apple's AVFoundation and the fact that I do not have an apple device to test with. I am working on this

morajabi, 2022-07-15:
@l1npengtul I'd be happy to test anything anytime. 
On a sidenote, what do you suggest that I can use today as a PoC for a WebRTC app I'm building until the new nokhwa is released?

l1npengtul, 2022-07-15:
OpenCV tends to work, if not a bit clunky.

Nokhwa has web support, so you may be interested in that.

bschwind, 2022-07-26:
@l1npengtul please let me know if you'd like any help testing or developing for AVFoundation stuff. I can't claim to be an expert, but I have written some bindings to VideoToolbox so I'm a bit familiar with the slight insanity required to write Rust bindings to Apple's frameworks.

l1npengtul, 2022-07-26:
@bschwind Yeah, I feel your pain. I am currently finishing the public API, so Ill keep this in mind!

morajabi, 2022-08-05:
@l1npengtul Thanks for your efforts, we're eagerly waiting for the new version!

morajabi, 2022-10-18:
Is the main branch in a working state for av foundation already? Let us know when you need testers!

l1npengtul, 2022-10-18:
I just need to finish up a local commit and I believe I can release an RC :D

closed 2022-11-05 by l1npengtul

l1npengtul, 2022-11-05:
This has been fixed in the lastest `senpai` by adding NV12 support for MacOS

bschwind, 2022-11-05:
Thanks so much, @l1npengtul!

commit 9da0c8fd (2022-11-05): add support for NV12 to fix #52, #51
commit 745d3f03 (2022-11-05): Add NV12 support for MacOS, fix #51, fix #52