l1npengtul/nokhwa #53: input-msmf fails to detect camera on windows

sayanarijit, 2022-07-20:
Hi, first of all, thanks for the awesome library. I published a project using this library: https://github.com/sayanarijit/qrscan which runs fine on Linux.

However, as [reported by a windows user](https://github.com/sayanarijit/qrscan/issues/7), it fails to detect the camera on windows.

I am using `input-msmf` for the windows build.

The error is:

```
error: qrscan: Could not get device property Devices: Not Found
```

Device info:

[image]

l1npengtul, 2022-07-20:
The screenshot attached is useless, please ask the user to provide the model of the webcam they are using,

Ginkooo, 2022-07-20:
I'm having the same issue too, on some windows 8 few years old asus

l1npengtul, 2022-07-20:
@Ginkooo have you tried changing your camera settings? Some cameras don't support YUYV/MJPEG, etc. I am working to fix this.

sayanarijit, 2022-07-21:
>  have 3 DirectShow webcams: [GoPro Webcam](https://community.gopro.com/s/article/GoPro-Webcam), OBS Virtual Camera, and [NVIDIA Broadcast](https://www.nvidia.com/en-us/geforce/broadcasting/broadcast-app/). And I have the [ManyCam Virtual Webcam](https://help.manycam.com/knowledge-base/webcam-driver/) for use in UWP applications.

> In a similar case, I have tried [zbar](https://github.com/mchehab/zbar) from MSYS2, and the ManyCam Virtual Webcam worked.

l1npengtul, 2022-11-06:
I honestly do not know why it doesn't work, I followed the Microsoft Documentation to a tee. Perhaps the camera just doesn't support MSMF.

closed 2022-11-06 by l1npengtul