l1npengtul/nokhwa #54: missing linkage to CoreVideo framework

jbg, 2022-08-25:
in `Cargo.toml`:

```
nokhwa = {
  version = "0.9", optional = true, default-features = false, features = [
    "decoding", "flume", "input-v4l", "input-msmf", "input-avfoundation", "output-threaded"
  ]
}
```

when running `cargo build`:

```
  = note: Undefined symbols for architecture arm64:
            "_CVPixelBufferLockBaseAddress", referenced from:
                _$LT$nokhwa_bindings_macos..avfoundation..CALLBACK_CLASS$u20$as$u20$core..ops..deref..Deref$GT$::deref::__static_ref_initialize::capture_out_callback::h96ec85fdab100000 in libnokhwa_bindings_macos-0e9400fda68c51e0.rlib(nokhwa_bindings_macos-0e9400fda68c51e0.nokhwa_bindings_macos.0da1d717-cgu.1.rcgu.o)
            "_CVPixelBufferGetPixelFormatType", referenced from:
                _$LT$nokhwa_bindings_macos..avfoundation..CALLBACK_CLASS$u20$as$u20$core..ops..deref..Deref$GT$::deref::__static_ref_initialize::capture_out_callback::h96ec85fdab100000 in libnokhwa_bindings_macos-0e9400fda68c51e0.rlib(nokhwa_bindings_macos-0e9400fda68c51e0.nokhwa_bindings_macos.0da1d717-cgu.1.rcgu.o)
            "_CVPixelBufferGetBaseAddress", referenced from:
                _$LT$nokhwa_bindings_macos..avfoundation..CALLBACK_CLASS$u20$as$u20$core..ops..deref..Deref$GT$::deref::__static_ref_initialize::capture_out_callback::h96ec85fdab100000 in libnokhwa_bindings_macos-0e9400fda68c51e0.rlib(nokhwa_bindings_macos-0e9400fda68c51e0.nokhwa_bindings_macos.0da1d717-cgu.1.rcgu.o)
            "_CVPixelBufferGetDataSize", referenced from:
                _$LT$nokhwa_bindings_macos..avfoundation..CALLBACK_CLASS$u20$as$u20$core..ops..deref..Deref$GT$::deref::__static_ref_initialize::capture_out_callback::h96ec85fdab100000 in libnokhwa_bindings_macos-0e9400fda68c51e0.rlib(nokhwa_bindings_macos-0e9400fda68c51e0.nokhwa_bindings_macos.0da1d717-cgu.1.rcgu.o)
            "_CVPixelBufferUnlockBaseAddress", referenced from:
                _$LT$nokhwa_bindings_macos..avfoundation..CALLBACK_CLASS$u20$as$u20$core..ops..deref..Deref$GT$::deref::__static_ref_initialize::capture_out_callback::h96ec85fdab100000 in libnokhwa_bindings_macos-0e9400fda68c51e0.rlib(nokhwa_bindings_macos-0e9400fda68c51e0.nokhwa_bindings_macos.0da1d717-cgu.1.rcgu.o)
          ld: symbol(s) not found for architecture arm64
          clang-11: error: linker command failed with exit code 1 (use -v to see invocation)
```

I think the fix would just be to add to `nokhwa-bindings-macos/build.rs`:

```
println!("cargo:rustc-link-lib=framework=CoreVideo");
```

morajabi, 2022-09-15:
@l1npengtul It has been added here: https://github.com/l1npengtul/nokhwa/commit/c090601599a312006b2f7f145b39e5f7c35c3d8a#diff-fcd41588568463ae306c393ffb28d6cb8425bb6bbc1b5b3c06472862c8caac07R21 
you can close it now.

l1npengtul, 2022-09-16:
Keeping this open until I can confirm fixed

l1npengtul, 2022-09-28:
Cannot recreate on latest commit. Closing.

closed 2022-09-28 by l1npengtul