l1npengtul/nokhwa #57: Platform requirements not satisfied. (No Selection)

heyarne, 2022-09-15:
I'm trying to use this library on NixOS, but I'm hitting the following error:

```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: NotImplementedError("Platform requirements not satisfied. (No Selection)")', src/main.rs:9:10
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: core::result::unwrap_failed
   3: core::result::Result<T,E>::unwrap
             at /build/rustc-1.62.1-src/library/core/src/result.rs:1078:23
   4: nannou_webcam::main
             at ./src/main.rs:5:22
   5: core::ops::function::FnOnce::call_once
             at /build/rustc-1.62.1-src/library/core/src/ops/function.rs:248:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace
```

I guess it's missing a dynamic library? My `Cargo.toml` looks like this:

```
[package]
name = "example"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
nokhwa = { git = "https://github.com/l1npengtul/nokhwa", rev = "0.9.4" }
```

And my `flake.nix` looks like this:

```
{
  description = "A very basic flake";

  inputs.nixpkgs.url = github:NixOS/nixpkgs;

  outputs = { self, nixpkgs }:
  let
    system = "x86_64-linux";
    pkgs = import nixpkgs { inherit system; };
    lib = pkgs.lib;
    buildInputs = with pkgs; [
      clippy
      rustc
      cargo
      cargo-watch
      rustfmt
      rust-analyzer

      # for nannou
      xorg.libX11
      xorg.libXcursor
      xorg.libXrandr
      xorg.libXi
      vulkan-tools vulkan-headers vulkan-loader vulkan-validation-layers
      # wayland
      # xwayland
    ];
  in {
    devShell.${system} = pkgs.mkShell {
      inherit buildInputs;
      RUST_BACKTRACE = 1;
    };
  };
}
```

Any help would be greatly appreciated. Thanks!

l1npengtul, 2022-09-15:
You need to enable "input-v4l" feature.

heyarne, 2022-09-15:
Thanks! The final `flake.nix` that made it work, including config for `rust-bindgen`, looks like this:

``` nix
{
  description = "A very basic flake";

  inputs.nixpkgs.url = github:NixOS/nixpkgs;

  outputs = { self, nixpkgs }:
  let
    system = "x86_64-linux";
    pkgs = import nixpkgs { inherit system; };
    lib = pkgs.lib;
    buildInputs = with pkgs; [
      clippy
      rustc
      cargo
      cargo-watch
      rustfmt
      rust-analyzer

      # nokhwa
      libclang
      linuxHeaders
      stdenv.cc.libc_dev
    ];
  in {
    devShell.${system} = pkgs.mkShell {
      inherit buildInputs;
      RUST_BACKTRACE = 1;

      LD_LIBRARY_PATH = (lib.makeLibraryPath buildInputs);
      BINDGEN_EXTRA_CLANG_ARGS = "-I${pkgs.linuxHeaders}/include -I${pkgs.stdenv.cc.libc_dev}/include";
    };
  };
}
```

closed 2022-09-15 by heyarne