l1npengtul/nokhwa #60: Can't use controls on windows

quartzo, 2022-10-10:
All requests to consult camera parameters (controls) are returning an error using input-msmf. I'm still trying to discover what is the problem...

I'm getting:
```
Err(GUIDSetError("MEDIA_FOUNDATION_FIRST_VIDEO_STREAM", "MF_MEDIASOURCE_SERVICE", "O objeto não oferece suporte ao serviço especificado."))
```

Probably from here:
```
pub fn control(&self, control: MediaFoundationControls) -> Result<MFControl, BindingError> {
            let media_source = unsafe {
                let mut receiver: MaybeUninit<IMFMediaSource> = MaybeUninit::uninit();
                let mut ptr_receiver = receiver.as_mut_ptr();
                if let Err(why) = self.source_reader.GetServiceForStream(
                    MEDIA_FOUNDATION_FIRST_VIDEO_STREAM,
                    &MF_MEDIASOURCE_SERVICE,
                    &IMFMediaSource::IID,
                    (&mut ptr_receiver as *mut *mut IMFMediaSource).cast::<*mut std::ffi::c_void>(),
                ) {
                    return Err(BindingError::GUIDSetError(
                        "MEDIA_FOUNDATION_FIRST_VIDEO_STREAM".to_string(),
                        "MF_MEDIASOURCE_SERVICE".to_string(),
                        why.to_string(),
                    ));
                }
                receiver.assume_init()
            };
```

l1npengtul, 2022-10-10:
Please include more information about your camera and setup.
Please also translate the error, I do not speak spanish.

quartzo, 2022-10-10:
It's Portuguese: "The object does not support the specified service."

I'm trying on Windows10 in two different computers and in 2 different webcams with the same results.
The first one is a Logitech C270 and the second one is a Brazilian PTZ Cam.
Both are returning and changing controls when used with the Linux backend (v4l2) on a different installation.

l1npengtul, 2022-10-10:
I have pushed an experimental fix to the 0.9.2 branch. Try it again using the git version please.

quartzo, 2022-10-10:
Got the same kind of errors...
I had to create a const to make it compile. I used Google to find the value, I hope it's correct:
```
    const MF_SOURCE_READER_MEDIASOURCE: u32 = 0xFFFF_FFFF;
```

quartzo, 2022-10-10:
There is no release method on the object, and the "into_raw" gives a hint:
"Returns the raw COM interface pointer and releases ownership. It the caller’s responsibility to release the COM interface pointer."
So, it seems that it works with only these changes.

l1npengtul, 2022-10-10:
You can release by using [IUnknown](https://microsoft.github.io/windows-docs-rs/doc/windows/core/struct.IUnknown.html).

l1npengtul, 2022-10-10:
I'll implement this tommorrow

closed 2022-10-12 by l1npengtul