l1npengtul/nokhwa #61: Saving frame fails with .save(), but not with image::save_buffer()?

paulmelis, 2022-10-10:
```
use nokhwa::*;

fn main() {
    let mut camera = Camera::new(
        0, 
        Some(CameraFormat::new_from(1280, 720, FrameFormat::MJPEG, 30)),
    )
    .unwrap();

    let frame_width = camera.resolution().width_x;
    let frame_height = camera.resolution().height_y;    

    camera.open_stream().unwrap();

    let updated_frame = camera.frame()
        .expect("could not get latest frame");

    // This works
    //image::save_buffer("frame.jpg", &updated_frame, frame_width, frame_height, image::ColorType::Rgb8)
    //    .expect("save_buffer");

    // This fails for some reason
    updated_frame.save("frame.jpg").unwrap();
}
```

With the latter direct `.save()` I get this error:
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Unsupported(UnsupportedError { format: Exact(Jpeg), kind: Format(Exact(Jpeg)) })', src/main.rs:24:37
```

I'm not exactly sure why that happens (don't know enough about Rust yet to figure that out)?

l1npengtul, 2022-10-10:
This is not a nokhwa issue. This is an issue with the image crate. Please open an issue there.

closed 2022-10-10 by l1npengtul

fintelia, 2022-10-10:
This is caused by _nokhwa_ importing _image_ with `default-features = false`. That setting cuts down on code size by stripping out support for encoding/decoding all the image formats causing them to return "Unsupported".