l1npengtul/nokhwa #71: 0.10.0-rc1 `output-threaded` no longer compiles

Ant59, 2022-11-17:
I am receiving the following errors when trying to compile a project which depends upon `nokhwa` on version `0.10.0-rc1`.

```
error[E0507]: cannot move out of a shared reference
  --> /home/antony/.cargo/registry/src/github.com-1ecc6299db9ec823/nokhwa-0.10.0-rc.1/src/threaded.rs:87:12
   |
87 |           Ok(*self
   |  ____________^
88 | |             .camera
89 | |             .lock()
90 | |             .map_err(|why| NokhwaError::GeneralError(why.to_string()))?
91 | |             .index())
   | |____________________^ move occurs because value has type `CameraIndex`, which does not implement the `Copy` trait
```

```
error[E0507]: cannot move out of a shared reference
   --> /home/antony/.cargo/registry/src/github.com-1ecc6299db9ec823/nokhwa-0.10.0-rc.1/src/threaded.rs:125:12
    |
125 |           Ok(*self
    |  ____________^
126 | |             .camera
127 | |             .lock()
128 | |             .map_err(|why| NokhwaError::GeneralError(why.to_string()))?
129 | |             .info())
    | |___________________^ move occurs because value has type `CameraInfo`, which does not implement the `Copy` trait
```

```
error[E0596]: cannot borrow `last_frame` as mutable, as it is not declared as mutable
   --> /home/antony/.cargo/registry/src/github.com-1ecc6299db9ec823/nokhwa-0.10.0-rc.1/src/threaded.rs:490:22
    |
489 |                 if let Ok(last_frame) = last_frame_captured.lock() {
    |                           ---------- help: consider changing this to be mutable: `mut last_frame`
490 |                     *last_frame = frame.clone();
    |                      ^^^^^^^^^^ cannot borrow as mutable
```

```
error[E0596]: cannot borrow `cb` as mutable, as it is not declared as mutable
   --> /home/antony/.cargo/registry/src/github.com-1ecc6299db9ec823/nokhwa-0.10.0-rc.1/src/threaded.rs:492:25
    |
491 |                     if let Ok(cb) = frame_callback.lock() {
    |                               -- help: consider changing this to be mutable: `mut cb`
492 |                         cb(frame);
    |                         ^^ cannot borrow as mutable

```

```
info: The currently active `rustc` version is `rustc 1.65.0 (897e37553 2022-11-02)`
```

Please let me know if any further information is needed.

l1npengtul, 2022-11-17:
Please try RC2 on Crates.io

Ant59, 2022-11-17:
This now compiles successfully. Thank you for your great work on this crate.

closed 2022-11-17 by Ant59