l1npengtul/nokhwa #75: AVFoundation: Channel closes after receiving one frame

gennyble, 2022-12-04:
I have a loop with that calls `Camera::frame_raw` at twice the FPS given by the `CameraInfo`. After receiving one frame every subsequent call results in "Could not capture frame: receiving on a closed channel". Finally the program dies with one of several faults. I think the unbounded channel is filling up and there's like, a stack collision?

It seems you're currently working on this part of the code. Feel free to ping me if you need me to test something. I'm currently on an M1 Macbook Air and would be happy to help where possible.

l1npengtul, 2022-12-04:
I pushed a commit to main. Try and see if it works.

The AVFoundation fix will be a part of the final release, along with a NV12 fix.

gennyble, 2022-12-04:
Unfortunately it didn't seem to work and the channel is still getting closed somehow.

gennyble, 2022-12-04:
Oh, hey! I got it to stop panicing! And chills because I got excited, too, lol.

I was looking around the docs for `Arc`. In `capture_out_callback` you use `Arc::from_raw`, but the docs say that you have to be sure that you only drop an `Arc` you've made that way only once; It seems that `Arc::into_raw` and `Arc::from_raw` are 1 to 1. Like, gotta call into for every from. So it looks like the sender is being dropped?

I added `std::mem::forget(buffer_sndr);` right after the if statement [here][code] and my repro is giving me, "Got frame!" more than once *(and in fact, seems to stop erroring)*.

I think with this you can go back to using `recv` instead of the `drain` for the receiver. It's still been giving me the heap things, anyway.

[code]: https://github.com/l1npengtul/nokhwa/blob/senpai/nokhwa-bindings-macos/src/lib.rs#L442

commit 32a33f64 (2022-12-04): Fix #75, thanks @gennyble
closed 2022-12-04 by l1npengtul