l1npengtul/nokhwa #82: Wrong colors in areas with high or low light

leofidus, 2022-12-22:
OS: Windows 11
Camera: `Name: AI Front Camera, Description: MediaFoundation Camera, Extra: \\?\display#qcom_avstream_8280_lenv#3&98d97ee&1&uid32768#{e5323777-f976-4f5b-9b55-b94699c46e44}\{4faeafd4-041b-4e46-85fd-400473891182}, Index: 0` (webcam of a Lenovo X13s)

I've been trying out the capture example, which I would assume to produce correct images out of the box. However, with my webcam I get correct colors in medium-light areas, but wrong colors in high or low light areas.

See the below two images, the first taken with ` cargo run -- single 0 img_test.png AbsoluteHighestResolution`, the second with Window's Camera app.

[image]
[image]

This looks like an HDR issue of some kind, but I'm not sure how to go about solving it.

l1npengtul, 2022-12-23:
Unfortunately, I do not have access to any HDR webcam(s).

benpoulson, 2023-01-17:
I don't believe it's related to HDR. 

Here's a photo taken from a [PS3 Webcam](https://en.wikipedia.org/wiki/PlayStation_Eye).
640×480 pixels - YUYV

[image]

```
Name: gspca main driver, Description: Video4Linux Device @ /dev/video3, Extra: , Index: 3
```

benpoulson, 2023-01-17:
[image]

Fixed it, it was bad conversion in the yuyv444_to_rgb function. I know you're doing major refactoring currently, but would you like a PR for that change?

l1npengtul, 2023-01-17:
@benpoulson That would be great, thank you! Sorry for the lack of progress, I'm dealing with some IRL issues at the moment 😅

yamt, 2023-10-17:
my fix for a similar symptom: https://github.com/l1npengtul/nokhwa/pull/150