l1npengtul/nokhwa #85: threaded.rs - 0.10.2 - "cannot find value `current_camera` in this scope"

JosephCatrambone, 2022-12-27:
When compiling a project with Nokhwa 0.10.2 and feature-threaded, I see the following build error:

Operating System: Windows 10 Pro
Cargo Version:  cargo 1.66.0 (d65d197ad 2022-11-15)
Rustc Version: stable-x86_64-pc-windows-gnu updated - rustc 1.66.0 (69f9c33d7 2022-12-12) (from rustc 1.64.0 (a55dd71d5 2022-09-19))

```
error[E0425]: cannot find value `current_camera` in this scope
   --> C:\Users\Joseph Catrambone\.cargo\registry\src\github.com-1ecc6299db9ec823\nokhwa-0.10.2\src\threaded.rs:112:13
    |
112 |             current_camera,
    |             ^^^^^^^^^^^^^^ a field by this name exists in `Self`

```

Looking at the relevant excerpt, it seems like this is true.  I don't see where current camera is set.

```
	pub fn with_custom(camera: Camera, callback: impl FnMut(Buffer) + Send + 'static) -> Self {
		CallbackCamera {
			camera: Arc::new(Mutex::new(camera)),
			frame_callback: Arc::new(Mutex::new(Box::new(callback))),
			last_frame_captured: Arc::new(Mutex::new(Buffer::new(
				Resolution::new(0, 0),
				&vec![],
				FrameFormat::GRAY,
			))),
			die_bool: Arc::new(Default::default()),
			current_camera,
			handle: Arc::new(Mutex::new(None)),
		}
	}
```

It seems like a quick patch to fix it, though I'm not sure I am familiar enough with the codebase to say if this will cause problems:

```
	pub fn with_custom(camera: Camera, callback: impl FnMut(Buffer) + Send + 'static) -> Self {
		let camera_info = camera.info().clone();
		CallbackCamera {
			camera: Arc::new(Mutex::new(camera)),
			frame_callback: Arc::new(Mutex::new(Box::new(callback))),
			last_frame_captured: Arc::new(Mutex::new(Buffer::new(
				Resolution::new(0, 0),
				&vec![],
				FrameFormat::GRAY,
			))),
			die_bool: Arc::new(Default::default()),
			current_camera: camera_info,
			handle: Arc::new(Mutex::new(None)),
		}
	}
```

I'll submit a patch, but wanted to have an associated issue just in case.

JosephCatrambone, 2022-12-27:
Just noticed this was fixed in the current ~main~ branch*.  https://github.com/l1npengtul/nokhwa/blob/senpai/src/threaded.rs#L103  

Sorry about the noise.

*Or should I say, "me noticed senpai"?

closed 2022-12-27 by JosephCatrambone

l1npengtul, 2022-12-28:
I forgot to publish this, Ill donit later tody