l1npengtul/nokhwa #90: Incorrect colors

br0kenpixel, 2023-01-08:
I'm using the following code to capture an image and save it to file:
```rust
fn capture_webcam() {
    let index = CameraIndex::Index(0);
    let requested = RequestedFormat::new::<RgbFormat>(RequestedFormatType::None);
    let mut camera = Camera::new(index, requested).unwrap();
    println!("{}", camera.info().human_name());

    camera.open_stream().expect("Permission denied");
    let frame = camera.frame().unwrap();
    camera.stop_stream().unwrap();
    let decoded = frame.decode_image::<RgbFormat>().unwrap();
    
    decoded.save("img.png").unwrap();
    println!("Captured webcam");
}
```
The result is a pink-ish image:
[image]

I tried:
- Changing `RgbFormat` to other formats
- Changing image file format from `png` to `jpeg`

Also, according to the documentation, `camera.open_stream()` should be called internally by `camera.frame()`, however, deleting the `camera.open_stream().expect("Permission denied");` line makes `camera.frame()` hang forever.

> macOS 12.6.1 (21G217), Intel
> nokhwa = { version = "0.10.3", features = ["input-native"] }
> image = "0.24.5"
> rustc 1.66.0

gennyble, 2023-01-08:
This *could* be due to #78. I tested the 0.10.3 code and it's still reporting as NV12 despite being successfully decoded as YUV422. I'm not sure, though, because `decode_image()` fails with "bad input buffer size". Just a lead, maybe?

yamt, 2023-10-17:
might be related: https://github.com/l1npengtul/nokhwa/pull/151