l1npengtul/nokhwa #92: arm support

makerio90, 2023-01-11:
is there any info on arm support? (`armv7`  and `aarch64`)?
id like to use this on a raspberry pi but I've had issues with cross compiling webcam lib's before.

if not, is there any plan to support arm architectures in the future?

l1npengtul, 2023-01-11:
There isnt anything inherent in the code that would prevent it from working on ARM. You will have to set up a C cross compiler though.

leofidus, 2023-01-11:
For what it's worth, 0.10.3 compiles and runs the examples on my aarch64 if I set `default-features = false` for the mozjpeg dependency in nokhwa-core (the purpose of that change is to disable he simd support of mozjpeg since that didn't compile for me)

makerio90, 2023-01-11:
nah, `aarch64` does not work with v4l, but thats an issue with `v4l-rs` not nokhawa

leofidus, 2023-01-12:
I was running Windows, so I can't comment on v4l. But still, you might have luck with the opencv backend?

l1npengtul, 2023-01-12:
I'll add this to the 0.11 roadmap...
Time to find a RPi... 😔

makerio90, 2023-01-12:
> I was testing on Windows, so I can't comment on v4l. But still, you might have luck with the opencv backend

ill try that for now
> I'll add this to the 0.11 roadmap... Time to find a RPi... :pensive:

no need, you can use a docker container with the `--platform` flag

l1npengtul, 2023-01-12:
Will add that to the docs... Ty!

makerio90, 2023-01-12:
...as in you can use docker to simulate a aarch64 machine, using a docker container is far from the best solution.

benpoulson, 2023-01-16:
Even easier, using [cross-rs](https://github.com/cross-rs/cross) you can just run
```bash
cross build --target "aarch64-unknown-linux-gnu " --release
```

UnknownError55, 2023-01-24:
@makerio90 I am currently running this library on an actual RaspberryPi 4 model B. The specifics of the OS is: `Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux` The only issue I have with it is getting the RaspberryPi HD Camera to work with the library. However, I have tested a Logitech webcam plugged into the RaspberryPi, and the library is able to pull out frames just fine.

makerio90, 2023-01-24:
OH! maybe it just has issues with cross compiles.

closed 2025-11-14 by l1npengtul