l1npengtul/nokhwa #93: fourcc for grayscale

wegmarken2006, 2023-01-11:
Hello, 
fourcc for grayscake should be "GREY" instead of "GRAY".

wegmarken2006, 2023-01-11:
I was not talking about the name used in the FrameFormat enum, which of course can be anything; I was talking about the string that is actually sent to the OS like in the write() below:
`
            FrameFormat::GRAY => {
                write!(f, "GRAY")
            }
`
 I don't know if I am missing something, I am using a yocto built linux, and the string "GRAY" is not recognized, see also https://www.kernel.org/doc/html/v4.10/media/uapi/v4l/pixfmt-grey.html

l1npengtul, 2023-01-11:
We don't follow convention of a single platform. FrameFormat is nokhwa's own way of handling 4CCs, and we don't plan on word for word compatibility with v4l2