l1npengtul/nokhwa #95: Examples are broken due to breaking changes

r4ve1, 2023-01-13:
Hi I'm new to both rust and nokhwa (pretty cool library 😸 by the way), so fixing the examples could really help me getting started with this library, thanks

Janik-Haag, 2023-07-12:
@r4ve1 did you write a appication with nokhaw yet? I tried to but got a bit frustrated with the example not working and then looked for issues and found this.

BonaBeavis, 2023-07-22:
```rust
// first camera in system
let index = CameraIndex::Index(0); 
// request the absolute highest resolution CameraFormat that can be decoded to RGB.
let requested = RequestedFormat::new::<RgbFormat>(RequestedFormatType::AbsoluteHighestFrameRate);
// make the camera
let mut camera = Camera::new(index, requested).unwrap();
camera.open_stream().unwrap(); // This line is missing
// get a frame
let frame = camera.frame().unwrap();
println!("Captured Single Frame of {}", frame.buffer().len());
// decode into an ImageBuffer
let decoded = frame.decode_image::<RgbFormat>().unwrap();
println!("Decoded Frame of {}", decoded.len());
```

r4ve1, 2023-07-23:
👏 Will try, thanks

closed 2023-07-23 by r4ve1