l1npengtul/nokhwa #98: Querying AVFoundation doesn't find all cameras

caerphoto, 2023-01-19:
I have a mirrorless camera connected to my MacBook Pro via a HDMI capture device that connects to a USB port. This camera isn't detected when I use AVFoundation backend; only the built-in FaceTime camera is detected:

```txt
❯ cargo run
   Compiling velut v0.1.0 (/Users/farrela/dev/personal/velut)
    Finished dev [unoptimized + debuginfo] target(s) in 1.49s
     Running `target/debug/velut`
User said true
0: CameraInfo { human_name: "FaceTime HD Camera (Built-in)", description: "Apple Inc.: UVC Camera VendorID_1452 ProductID_34068 - AVCaptureDeviceTypeBuiltInWideAngleCamera, Unspecified f0", misc: "0x8020000005ac8514", index: Index(0) }
1280x720 3686400 naripoggers
1280x720 3686400 callback thread
(etc.)
```

If I run `ioreg | grep -i cam` on the command line it only shows the FaceTime camera:
```txt
❯ ioreg | grep -i cam
    | |   |   |   | | +-o FaceTime HD Camera (Built-in)@80200000  <class IOUSBHostDevice, id 0x100000507, registered, matched, active, busy 0 (314 ms), retain 25>
    | |   | +-o FaceTime HD Camera (Built-in)@80200000  <class AppleUSBDevice, id 0x100000543, registered, matched, active, busy 0 (1 ms), retain 13>
```

But if I run `system_profiler SPCameraDataType` it shows both cameras:

```txt
❯ system_profiler SPCameraDataType
Camera:

    FaceTime HD Camera (Built-in):

      Model ID: UVC Camera VendorID_1452 ProductID_34068
      Unique ID: 0x8020000005ac8514

    USB Camera VID:7758 PID:28694:

      Model ID: UVC Camera VendorID_7758 ProductID_28694
      Unique ID: 0x11110001e4e7016
```

I don't really know ~~enough~~ anything about AVFoundation, sadly, otherwise I'd offer to help fix this, although judging by the comments in /nokhwa-bindings-macos)/src/lib.rb, perhaps this is for the best.

bookshiyi, 2023-01-31:
Theoretically, this problem has already been solved by  https://github.com/l1npengtul/nokhwa/pull/83. 
But I found that the code on my computer which pull from crate.io is still not up-to-date, and I can only manually modify the misspelling to `AVCaptureDeviceTypeExternalUnknown` in my local.

l1npengtul, 2023-02-04:
Ill push a revision today.

hadhoryth, 2023-04-17:
@l1npengtul I tried to install the latest version of the package but it seems like the code is the same and I can not see the camera.  

Can not build from source. Maybe you can help with it? this is my `cargo.toml`

```toml
nokhwa = "0.10.3"
## Then tried this one which is failing
nokhwa = {git = "https://github.com/l1npengtul/nokhwa", branch="senpai"}
```

olonho, 2023-05-14:
Same issue here. Would it be possible to release fixed 0.10.x or 0.11, as this issue is quite major one?

l1npengtul, 2023-05-14:
0.10.4 has been published to fix this.

commit c46fb4be (2023-05-14): try and fix #98 again