l1npengtul/nokhwa #99: Cannot use on RaspberryPi's HD camera

UnknownError55, 2023-01-23:
I consistently get the error: `GetPropertyError { property: "CameraFormat", error: "Failed to Fufill" }` while trying to create a camera instance on the RaspberryPi's HD camera. Regardless if I attempt to use Rgb, RgbA, Luma, or LumaA as decoder method with `AbsoluteHighestFrameRate`. I decided to step through the program with `gdb` to see why it thought nothing was available, and I get quite a few formats available:
[image]
But right below, there is a section that filters this list down on what is functionally possible with the library, marked with the following comment: `// i write unmaintainable blobs of code because i am so cute uwu~~`. And unfortunately, this is where `gdb` fails me, or at least my experience with `gdb`, as it is very tricky to debug that large section of iterators iterating iterators.

Is there an easier way to correlate why these formats are being rejected by this library, or which modes are available? Additionally, the error message "Failed to Fufill", fulfill is misspelled.

UnknownError55, 2023-01-24:
Specifically, this seems to be a problem with the Raspberry Pi HD Camera (`bm2835 mmal`). As I can plug in a spare USB camera, for testing I used a Logitech HD Pro Webcam C920, and I am able to pull frames out no problem. So running this library on a RaspberryPi with aarch64 appears to be not a problem.

With this in mind, I believe that there might be a bit of an issue with this library trying to convert a colorspace, encoding or other configuration for this camera preventing this library from finding a configuration capable of accepting this camera. As I am not entirely sure what this library is currently capable of, and see that a major rework is currently in progress with the `senpai` branch, I am not sure if this issue is being remedied in the new set of features to eventually be rolled out. Or simply that this camera is too obscure to produce features for.

To assist in seeing what this camera can offer, I ran `v4l2-ctl -d /dev/video0 --all` for the RaspberryPi's HD Camera (`/dev/video0`) to see what it offers:
```
Driver Info:
        Driver name      : bm2835 mmal
        Card type        : mmal service 16.1
        Bus info         : platform:bcm2835-v4l2-0
        Driver version   : 5.15.84
        Capabilities     : 0x85200005
                Video Capture
                Video Overlay
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x05200005
                Video Capture
                Video Overlay
                Read/Write
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 0: ok)
Format Video Capture:
        Width/Height      : 1024/768
        Pixel Format      : 'JPEG' (JFIF JPEG)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 786432
        Colorspace        : JPEG
        Transfer Function : Default (maps to sRGB)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
        Flags             :
Format Video Overlay:
        Left/Top    : 150/50
        Width/Height: 1024/768
        Field       : None
        Chroma Key  : 0x00000000
        Global Alpha: 0xff
                Clip Count  : 0
        Clip Bitmap : No
Framebuffer Format:
        Capability    : Extern Overlay
                        Global Alpha
        Flags         : Overlay Matches Capture/Output Size
        Width         : 1024
        Height        : 768
        Pixel Format  : 'YU12'
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30000/1000)
        Read buffers     : 1

User Controls

                     brightness 0x00980900 (int)    : min=0 max=100 step=1 default=50 value=50 flags=slider
                       contrast 0x00980901 (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                     saturation 0x00980902 (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                    red_balance 0x0098090e (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                   blue_balance 0x0098090f (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                horizontal_flip 0x00980914 (bool)   : default=0 value=0
                  vertical_flip 0x00980915 (bool)   : default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=3 default=1 value=1
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
[... 97 lines cut ...]
                                10: Greyworld
            image_stabilization 0x009a0916 (bool)   : default=0 value=0
                iso_sensitivity 0x009a0917 (intmenu): min=0 max=4 default=0 value=0
                                0: 0 (0x0)
                                1: 100000 (0x186a0)
                                2: 200000 (0x30d40)
                                3: 400000 (0x61a80)
                                4: 800000 (0xc3500)
           iso_sensitivity_auto 0x009a0918 (menu)   : min=0 max=1 default=1 value=1
                                0: Manual
                                1: Auto
         exposure_metering_mode 0x009a0919 (menu)   : min=0 max=3 default=0 value=0
                                0: Average
                                1: Center Weighted
                                2: Spot
                                3: Matrix
                     scene_mode 0x009a091a (menu)   : min=0 max=13 default=0 value=0
                                0: None
                                8: Night
                                11: Sports

JPEG Compression Controls

            compression_quality 0x009d0903 (int)    : min=1 max=100 step=1 default=30 value=30
```

UnknownError55, 2023-01-24:
I just verified that the alternate Logitech webcam does work on the RaspberryPi using this library. The output image from a test application to test the Logitech webcam on the RaspberryPi using this library are reasonably color accurate (accurate by eye, no egregious color inaccuracies). So it does appear to simply be that the library is just not able to accept the RaspberryPi HD Camera's available capabilities.
I know that gathering obscure webcams and equipment to test can be rather prohibitive. So, if you need, I can run additional commands for you to interrogate the camera for available features or run code to test various features of the camera. Let me know.

l1npengtul, 2023-01-24:
You're in luck, I was going to do a project with a raspberry pi camera. I'll get to you sometime later.

timfpark, 2023-02-23:
I can confirm.  With a Raspberry Pi Zero and the [Adafruit Zero Camera](https://www.adafruit.com/product/5389) I am seeing the same "Failed to Fufill" error.

michivi, 2023-05-29:
Had the same issue. Fixed it with a PR #121 and ensuring the following were configured in the `/boot/config.txt` file:

```text
dtoverlay=vc4-fkms-v3d
gpu_mem=128
start_x=1
```

The library now works on my Raspberry Pi 3 with a Raspberry Pi camera.