l1npengtul/nokhwa #100: `camera.info`'s resolution is different from number of `camera.frame`'s slice elements

bokutotu, 2023-01-24:
Thank you for creating such a great repository!
I have a issue when using Mabbook pro 14inc (M1 max).
Below are the steps to reproduce the error.

Ventura 13.1
machine MacBook Pro 14-inch 2021 Apple M1 Max
rust version 1.66.0
Cargo.toml
```
name = "show_image"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
show-image="0.13.1"
image = { version="0.24.5" }
nokhwa = { version = "0.10.3", features=["input-native", "output-threaded"] }
```
Code (`main.rs`)
```rust
use nokhwa::{
    nokhwa_initialize,
    pixel_format::{RgbAFormat, RgbFormat},
    query,
    utils::{ApiBackend, RequestedFormat, RequestedFormatType},
    threaded::CallbackCamera
};

fn main() {
    // only needs to be run on OSX
    nokhwa_initialize(|granted| {
        println!("User said {}", granted);
    });
    let cameras = query(ApiBackend::Auto).unwrap();
    cameras.iter().for_each(|cam| println!("{:?}", cam));

    let format = RequestedFormat::new::<RgbFormat>(RequestedFormatType::AbsoluteHighestFrameRate);

    let first_camera = cameras.first().unwrap();

    let mut threaded = CallbackCamera::new(first_camera.index().clone(), format, |buffer| {
        let image = buffer.decode_image::<RgbAFormat>().unwrap();
        println!("{}x{} {}", image.width(), image.height(), image.len());
    })
    .unwrap();
    threaded.open_stream().unwrap();
    #[allow(clippy::empty_loop)] // keep it running
    loop {
        let frame = threaded.poll_frame().unwrap();
        let image = frame.decode_image::<RgbAFormat>().unwrap();
        println!(
            "{}x{} {} naripoggers",
            image.width(),
            image.height(),
            image.len()
        );
    }
}
```
Error

```
User said true
CameraInfo { human_name: "FaceTime HD Camera", description: "Apple Inc.: FaceTime HD Camera - AVCaptureDeviceTypeBuiltInWideAngleCamera, Unspecified f0", misc: "47B4B64B-7067-4B9C-AD2B-AE273A71F4B5", index: Index(0) }
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ProcessFrameError { src: NV12, destination: "RGB", error: "bad input buffer size" }', src/main.rs:63:56
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: ProcessFrameError { src: NV12, destination: "RGB", error: "bad input buffer size" }', src/main.rs:55:57
```

bookshiyi, 2023-01-31:
Source frame format NV12 may not be supported on macintosh untill now, I guess.

l1npengtul, 2023-01-31:
It is likely a fault of the library improperly setting FourCC

hallucinogen, 2023-03-19:
I found that the first problem is the NV12 checker in this line: https://github.com/l1npengtul/nokhwa/blob/ad93e0a6c521f42424e7ec717455eece9f3e1fd2/nokhwa-core/src/types.rs#L1445

```
if data.len() != ((resolution.width() * resolution.height() * 3) / 2) as usize {
```

I tried to change it into

```
if data.len() != (resolution.width() * resolution.height() * 2) as usize {
```

Now it passes the decoding round.

But the resulting decoded image is faulty. It looks like it's split in the middle, and then we put the left part on the right and the right part on the left. I feel I am close to a solution, but can't really point it out

[image]

vaqxai, 2023-03-31:
Does not work on Macbook Air 2014 13". Same error.

stolinski, 2023-06-21:
Anyone have a work around for this? All cams in macOS are coming in as NV12 and hitting the same `bad input buffer size issue`.

anselanza, 2023-06-22:
Damn, I just hit this problem, too.

Until NV12 support comes in v0.11, it's possible to use a crate like [DCV Color Primitives](https://crates.io/crates/dcv-color-primitives) to do the necessary conversion.

bluezheng, 2023-07-17:
Hi, is there any workaround for this issue?

marcpabst, 2023-10-03:
Hi, I'm seeing this issue as well. I think the problem is that AVfoundation returns frames in `UYVY` format (https://wiki.videolan.org/YUV#UYVY), even if you tell nokhwa to request `NV12` (or some other format).

Here is my current workaround to deal with this kind of data (note that this only works on Rust nightly):

```rust
#![feature(portable_simd)]
use std::simd::SimdFloat;
use std::simd::f32x4;
use rayon::prelude::*;

#[inline]
pub fn uyvy_to_rgb24(in_buf: &[u8], out_buf: &mut [u8]) {
    debug_assert!(out_buf.len() as f32 == in_buf.len() as f32 * 1.5);

    in_buf
        .par_chunks_exact(4) // FIXME: use par_array_chunks() when stabalized (https://github.com/rayon-rs/rayon/pull/789)
        .zip(out_buf.par_chunks_exact_mut(6))
        .for_each(|(ch, out)| {
            let y1 = ch[1];
            let y2 = ch[3];
            let cb = ch[0];
            let cr = ch[2];

            let (r, g, b) = ycbcr_to_rgb(y1, cb, cr);

            out[0] = r;
            out[1] = g;
            out[2] = b;

            let (r, g, b) = ycbcr_to_rgb(y2, cb, cr);

            out[3] = r;
            out[4] = g;
            out[5] = b;
        });
}

// COLOR CONVERSION: https://stackoverflow.com/questions/28079010/rgb-to-ycbcr-using-simd-vectors-lose-some-data

#[inline]
fn ycbcr_to_rgb(y: u8, cb: u8, cr: u8) -> (u8, u8, u8) {
    let ycbcr = f32x4::from_array([y as f32, cb as f32 - 128.0f32, cr as f32 - 128.0f32, 0.0]);

    // rec 709: https://mymusing.co/bt-709-yuv-to-rgb-conversion-color/
    let r = (ycbcr * f32x4::from_array([1.0, 0.00000, 1.5748, 0.0])).reduce_sum();
    let g = (ycbcr * f32x4::from_array([1.0, -0.187324, -0.468124, 0.0])).reduce_sum();
    let b = (ycbcr * f32x4::from_array([1.0, 1.8556, 0.00000, 0.0])).reduce_sum();

    (clamp(r), clamp(g), clamp(b))
}

#[inline]
fn clamp(val: f32) -> u8 {
    if val < 0.0 {
        0
    } else if val > 255.0 {
        255
    } else {
        val.round() as u8
    }
}
```

As a sidenote, it looks like recent MacOS versions do not support transparent raw access to the camera outputs anymore, i.e. you will always be offered `uyvy422`, `yuyv422`, `nv12`, `0rgb`,  and `bgr0`, regardless of what the camera actually supports. MJPEG streams will still work when requesting a resolution/fps mode only supported through MJPEG (as is commonn for higher resolutions on UVC cameras) , but AVfoundation will handle the conversion to one of the listed pixel formats internally. 

@l1npengtul, not sure what the current status is here (I'm a bit confused with the different branches) but I might be able to help you with debugging this.

yamt, 2023-10-17:
> Hi, I'm seeing this issue as well. I think the problem is that AVfoundation returns frames in UYVY format (https://wiki.videolan.org/YUV#UYVY), even if you tell nokhwa to request NV12 (or some other format).

see https://github.com/l1npengtul/nokhwa/pull/151

l1npengtul, 2024-06-27:
Honestly, the current avfoundation code is in such a state that Ive blanked it out of my mind. I really cant help you here, unfortunately. 

I'll work on making nokhwa itself use avfoundation.

raphaelmenges, 2024-07-02:
> Just change
> `let [v, y2, u, y1] = [pixel[0] as f32, pixel[1] as f32, pixel[2] as f32, pixel[3] as f32];`
> to
> `let [u, y1, v, y2] = [pixel[0] as f32, pixel[1] as f32, pixel[2] as f32, pixel[3] as f32];`
> so you can get the image with the correct color.

Thanks @Moon1102 for the proposed workaround. It works on my MacBook with M1 and 14.5!

paviro, 2024-07-04:
@Moon1102 using your code I still get this result:
[image]
Any idea? This is a USB capture card connected to the Mac itself.

paviro, 2024-07-05:
Thank you for the tips @Moon1102 will investigate further!

paviro, 2024-07-09:
@Moon1102 My camera also seems to output 2vuy.  Not sure what I am doing wrong? Could you give me any more hints? Totally fine if you don't want to sadly I am a bit new to rust and the video world.
```
pixel buffer: "<CVPixelBuffer 0x600001c8f4f0 width=1920 height=1080 bytesPerRow=3840 pixelFormat=2vuy iosurface=0x6000029ecf10 surfaceid=2146 attributes={
    Height = 1080;
    IOSurfaceProperties =     {
        IOSurfacePurgeWhenNotInUse = 1;
    };
    PixelFormatType = 846624121;
    Width = 1920;
} propagatedAttachments={
    CGColorSpace = "<CGColorSpace 0x600000fed3e0> (kCGColorSpaceICCBased; kCGColorSpaceModelRGB; Composite NTSC)";
    CVImageBufferColorPrimaries = "SMPTE_C";
    CVImageBufferTransferFunction = "ITU_R_709_2";
    CVImageBufferYCbCrMatrix = "ITU_R_601_4";
} nonPropagatedAttachments={
}>"
```

paviro, 2024-07-10:
I don't fully understand as you can see the av-foundation binding does return 846624121 but somehow the same camera in nokhwa seems to return 875704438. Anyhow glad this works for now! Thank you for your support!

shadrach-tayo, 2024-10-23:
> > @Moon1102 My camera also seems to output 2vuy. Not sure what I am doing wrong? Could you give me any more hints? Totally fine if you don't want to sadly I am a bit new to rust and the video world.
> > ```
> > pixel buffer: "<CVPixelBuffer 0x600001c8f4f0 width=1920 height=1080 bytesPerRow=3840 pixelFormat=2vuy iosurface=0x6000029ecf10 surfaceid=2146 attributes={
> >     Height = 1080;
> >     IOSurfaceProperties =     {
> >         IOSurfacePurgeWhenNotInUse = 1;
> >     };
> >     PixelFormatType = 846624121;
> >     Width = 1920;
> > } propagatedAttachments={
> >     CGColorSpace = "<CGColorSpace 0x600000fed3e0> (kCGColorSpaceICCBased; kCGColorSpaceModelRGB; Composite NTSC)";
> >     CVImageBufferColorPrimaries = "SMPTE_C";
> >     CVImageBufferTransferFunction = "ITU_R_709_2";
> >     CVImageBufferYCbCrMatrix = "ITU_R_601_4";
> > } nonPropagatedAttachments={
> > }>"
> > ```
> 
> @paviro Sorry, I just saw the message and found that you've already solved the problem in a simpler and better way. That's awesome. I debugged it and got a format value of 846624121, which is different from yours. If your value is 875704438, that corresponds to [kCVPixelFormatType_420YpCbCr8BiPlanarVideoRange](https://github.com/libark/core-video-rs/blob/fd4a2576080d3e1bd3f92f787c4cd7add0963293/src/pixel_buffer.rs#L65) and it doesn't matter anymore.

Running into the same issue and I've applied the same patch, I get the same pixel format type `875704438` when using nokhwa, but when I decode to jpeg I still get this
[image]

so I applied this function you posted here https://github.com/l1npengtul/nokhwa/issues/100#issuecomment-2194272291 to the raw buffer and saved it as jpg and got this dark image under bright light
[image]

I'm on a M3 pro max btw, I'd appreciate it If anyone can help in anyway @paviro  @Moon1102 @l1npengtul

l1npengtul, 2024-10-23:
I'll see what I can do, but this will likely be fixed in 0.11 since we are switching to the avfoundation bindings crate instead of the current error prone msg send implementation.

l1npengtul, 2024-10-23:
It also doesn't help I don't have a mac to test this with... up to this point mac support was mostly me Winging It(tm)

shadrach-tayo, 2024-10-23:
> I'll see what I can do, but this will likely be fixed in 0.11 since we are switching to the avfoundation bindings crate instead of the current error prone msg send implementation.

Thanks for your good work 🙏🏽, Is there any temporary solution around this? It's been kicking my ass for a while now. I even tried ffmpeg to raw image conversion and still got the same result.

How soon is 0.11 going to be out? I'd love to help anyway I can too

shadrach-tayo, 2024-10-23:
when I set the capture device to my iPhone camera I get a clear, blurred or dark image sometimes
[image]
[image]

I suspect the camera/video setting device configuration also can be an issue on Macos/Iphone

l1npengtul, 2024-10-23:
> Thanks for your good work 🙏🏽, Is there any temporary solution around this? It's been kicking my ass for a while now. I even tried ffmpeg to raw image conversion and still got the same result.

Its probably AVFoundation backend and improperly set framebuffer settings. 

> How soon is 0.11 going to be out? I'd love to help anyway I can too

Follow tracking issue #86 

> I suspect the camera/video setting device configuration also can be an issue on Macos/Iphone

I suspect the blurry image is due to auto focus.

astraw, 2024-12-23:
I followed the idea from @paviro above and made a PR with a fix (#193) that works on my M1 Macbook Pro. You can use it already by using this in your `[dependencies]` in Cargo.toml:

```
nokhwa = { git = "https://github.com/astraw/nokhwa", rev = "0d09cc32c6bd06325115b2542cf58afc6f9eca8f", features = [
    "input-native",
] }
```

PR #193 (2024-12-23): macOS: use YUYV for certain pixel types
On macOS, use YUYV for the following pixel types:

- `kCVPixelFormatType_420YpCbCr10BiPlanarVideoRange`
- `kCVPixelFormatType_420YpCbCr8BiPlanarFullRange`
- `kCVPixelFormatType_420YpCbCr8BiPlanarVideoRange`

Fixes #100, at least for me (M1 Macbook Pro).

This was suggested by the work of @paviro at
https://github.com/l1npengtul/nokhwa/issues/100#issuecomment-2217577611. (Thanks!)

@l1npengtul this is a small patch on the 0.10 branch. Any chance you could merge it and make a quick 0.10.6 release in the meantime before 0.11 arrives? I have been testing thi [... cut]
commit 45c6a3ae (2024-12-23): macos: use YUYV for pixel types (#193)

kCVPixelFormatType_420YpCbCr10BiPlanarVideoRange
kCVPixelFormatType_420YpCbCr8BiPlanarFullRange
kCVPixelFormatType_420YpCbCr8BiPlanarVideoRange

Fixes #100.

This was suggested by the code from @paviro at
https://github.com/l1npengtul/nokhwa/issues/100#issuecomment-2217577611.