l1npengtul/nokhwa #101: Nokhwa hangs during camera init on windows with the "Cam Link 4k" by elgato.

ConnorBP, 2023-01-29:
I have a video feed at 1080p 60 going into my elgato cam link 4k which i have also tried at 30 fps and it seems to pause somewhere during init when i call this: `let mut camera = Camera::new(index,requested)?;`. The init runs smoothly for my built in web cam, but for whatever reason when i change index to 1 which is my cam link it freezes without error and hangs forever. Not quite sure what is going on here just yet but it is quite annoying and almost certainly a bug.

Giotino, 2023-10-01:
Hi, I found out that it also happens with my Logitech HD Pro Webcam C920.
With some debugging I found out that an `index += 1;` was missing from the MediaFoundation "drivers" for Windows.

Looking at the current `senpai` (`master`, `main`) branch it has been fixed:
https://github.com/l1npengtul/nokhwa/blob/dbdb42bdc6c377b029cf40b3cca02c0a76cedf53/nokhwa-bindings-windows/src/lib.rs#L643
but the current release `0.10.3` has been released 3 day before the fix... (https://github.com/l1npengtul/nokhwa/commit/c6ddc22d202b950d52ddac34aaafbde44dac48af)
 
@l1npengtul
Trying to apply that patch to the `0.10.3` version fixes the issue.
With that I found out there's another issue: the `compatible_format_list` contains a lot of duplicates, in my case 1136 total formats when there are only 284 unique ones.

I think it can be easly fixed by deduping the `framerate_list` that contains [0, 0, 0, X, X, X, X] (where X is the framerate).
The `framerate_list` is also initialized with three 0es, I'm not sure why.

l1npengtul, 2023-10-01:
i was probably sleep deprived when i wrote that

Giotino, 2023-10-01:
I don't know the release cycle of this crate, but since it affects some of the most popular webcam on the market (and not only some unknown brands) and since the last (and only) update from the release of the 0.10.3 (and 0.10.4) version was the one cited in my previous message: wouldn't be a good idea to release a `nokhwa-bindings-windows 0.4.1` with this update (and maybe the deduplication of formats)?

BTW: the unrecognized format that broke my camera (and probably @ConnorBP 's one) was H.264 which I immagine is offered by a lot of known brands and medium to high end quality webcams.

l1npengtul, 2023-10-01:
yeah, I'll release a windows 0.4.1

h264 wont be until 0.11 though :c

natisitotaw, 2024-05-10:
it still get stuck in let mut camera = Camera::new(index,requested)? when using my  Logitech HD Pro Webcam C920.

Gaming32, 2025-07-05:
The issue appears to be fixed in the repo, however, version 0.4.2 on crates.io is completely different from the current version in the repo, which seems very odd.

Gaming32, 2025-07-05:
The issue is fixed on the senpai branch, however, the issue is still present on the 0.10 branch, which is what is on crates.io.

Gaming32, 2025-07-05:
Or more specifically, it is fixed on the 0.10 branch, but the latest 0.10 branch has not been published to crates.io.