l1npengtul/nokhwa #102: Can't find any camera in windows 11

ololoshka2871, 2023-02-05:
I created a small application, that captures an image from an OBS virtual camera, processes it and sends it to my device via USB. Everything works fine in linux, but I decided to try to run it in Windows 11 and failed! The application cannot find any camera devices.

At first I decided that the problem is in my code and ran the "capture" example, but it also does not work. Is the problem related to MSMF in Windows 11?

Other applications interact with OBS Camera normally.

Rust: 1.67.0
nokwa: 0.10.3

[image]

l1npengtul, 2023-02-05:
It is possible that OBS doesn't produce a MSMF device, I'll have a look on this later.

Scarjit, 2023-05-13:
I have written a simple program to detect DirectShow and MF source (https://github.com/Scarjit/windows-camera-list), and it indeed lists OBS as DirectShow:

[image]

l1npengtul, 2023-06-15:
This is a wontfix - I do not have the support, time, or motivation to write another backend. If anyone is willing to write a DirectShow backend that would be nice, but atm any windows support outside of MSMF is not planned.