l1npengtul/nokhwa #103: RaspberryPi pulls frames at half frame rate

UnknownError55, 2023-02-19:
Due to the [RaspberryPi HD camera not working](#99), I decided to see if I could move ahead and use a Logitech webcam I had laying around in the RaspberryPi. Specifically the Logitech HD Pro Webcam C920. I went ahead to use that webcam in conjunction with [rs-webrtc](https://github.com/webrtc-rs/webrtc) to see if I could stream the webcam with low latency. I got some rather interesting results. So I made a test application below to test some of my findings:

```rs
use nokhwa::{
    nokhwa_initialize,
    pixel_format::RgbFormat,
    threaded::CallbackCamera,
    utils::{CameraIndex, RequestedFormat, RequestedFormatType},
};
use std::time::Instant;
use std::{thread, time};

fn main() {
    // Only needs to be run on OSX
    nokhwa_initialize(|granted| {
        println!("User said {}", granted);
    });

    let format = RequestedFormat::new::<RgbFormat>(RequestedFormatType::AbsoluteHighestFrameRate);

    let mut last = Instant::now();
    let index = CameraIndex::Index(0);
    let mut threaded = CallbackCamera::new(index, format, move |_buffer| {
        let now = Instant::now();
        let dt = now - last;
        println!("DT = {:?}; {} fps", dt, (1.0 / dt.as_secs_f64()).round());
        last = now;
    })
    .unwrap();

    threaded.open_stream().unwrap();

    println!("Camera FPS: {}", threaded.frame_rate().unwrap());

    #[allow(clippy::empty_loop)] // Keep it running
    loop {
        thread::sleep(time::Duration::from_millis(100));
    }
}
```

On the RaspberryPi with the Logitech webcam, `video0` -> YUYV422 and `video1` -> MJPG. Or at least this is what I gather from:
```
$ v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
        [1]: 'MJPG' (Motion-JPEG, compressed)
```
Trying to run off `video1` I get `GetPropertyError { property: "CameraFormat", error: "Failed to Fufill" }`. So I guess MJPG streams are not currently supported?

When running this on my laptop, I get that the camera can support 30fps and I get 30fps. However, on the RaspberryPi, the camera specifies that it can provide 30fps, but really, only provides 15fps.

Note that I removed any logic to actually decode frames themselves to just focus on the cyclic mechanism to fetch frames. Still, the RaspberryPi consistently has half the frame rate that was specified for the webcam.

PS.
I saw that there is quite a bit of work on the upcoming `0.11` version (senpai branch) to heavily revamp on, well, the everything. Specifically `FrameFormat`. Which introduces quite a bit new additions. Is it the idea that this library will contain capabilities to convert to not only from cameras with a wide variety of frame formats, but to also encode into any specified frame format? Such as encoding from YUYV422 to something like H265 or VP9? Or is the idea just to dump to RGB/RGBA from whichever format the camera provides?

l1npengtul, 2023-02-19:
First, make sure you're not decoding on the same thread as polling the camera. 

To answer your `FrameFormat` question, I don't plan to add them, simply add them as "ghost formats" so users can add their own decoding implementation. The goal is to make nokhwa more pluggable.

UnknownError55, 2023-02-19:
> First, make sure you're not decoding on the same thread as polling the camera.
The code snippet above exhibits the same behavior and it does not perform any decoding. The snippet only pulls frames and throws away the buffer it receives. The main thread enters an infinite loop that sleeps forever after the camera is created simply to let the threaded camera loop do its thing.

l1npengtul, 2023-02-19:
Can you post the full result of `v4l2-ctl --list-formats-ext`?

UnknownError55, 2023-02-19:
```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x90
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 432x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 640x360
[... 163 lines cut ...]
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1600x896
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
```

UnknownError55, 2023-02-19:
Also, when I run the snippet from this issue, the line `println!("Camera FPS: {}", threaded.frame_rate().unwrap());` prints out that it is trying to use 30fps.

l1npengtul, 2023-02-19:
Is the camera in a bright environment? That is sometimes an issue.

UnknownError55, 2023-02-19:
That is interesting. So I guess the framerate slows down to compensate for the exposure in dark environments? And I am guessing that is more on the V4L side doing the auto adjust?

l1npengtul, 2023-02-19:
Yes, it does that sometimes.

closed 2023-02-20 by UnknownError55

RReverser, 2023-04-21:
> I wonder if there's any way to retrieve what was the actual frame exposure duration...

Ah, I guess since `.frame()` is a blocking call, exposure time should be pretty much same as duration of the call itself.