l1npengtul/nokhwa #104: How to detect, if camera is open on windows.

thinkrapido, 2023-02-20:
I have this here     

```
let mut is_streaming = false;
        for idx in 0..4u32 {
            if let Err(NokhwaError::SetPropertyError{..}) = Camera::new(nokhwa::utils::CameraIndex::Index(idx), RequestedFormat::new::<LumaFormat>(nokhwa::utils::RequestedFormatType::None)) {
                is_streaming = true;
                break;
            }
        }
   ```
This works pretty good for linux, but I can't detect if the camera is open/streaming/on with any specific desktop app.

Can someone help me out, or is ther no functionality for this in nokhwa?

bookshiyi, 2023-02-23:
@thinkrapido You can try it like this
```
fn available_cameras() -> Result<Vec<CameraInfo>, NokhwaError> {
    match nokhwa::query(ApiBackend::Auto) {
        Ok(devices) => {
            println!("Cameras found: {:?}", devices.len());
            devices.iter().for_each(|x| {
                camera_info_print(x.clone());
            });
            if devices.is_empty() {
                return Err(NokhwaError::NotImplementedError(
                    "No cameras found".to_string(),
                ));
            }
            Ok(devices)
        }
        Err(e) => Err(e),
    }
}
```

OlivierLDff, 2023-06-07:
You need to start streaming on windows, since mf backend allow multiple camera handle to point to the same internal camera. Only one handle can open streaming