l1npengtul/nokhwa #110: Returns incorrect frame rate on Windows

RReverser, 2023-04-16:
Nokhwa seems to correctly detect camera formats for my integrated webcam, including framerate, but when retrieving it directly from created `Camera`, it always returns `1`.

Here's what the built-in demo returns as list of formats (and I can reproduce that all formats are correctly retrieved in my own Rust code as well):

```
C:\Users\me\Documents\nokhwa\examples\capture [(0.10.3)]> cargo run -- list-properties 0 All
   Compiling nokhwa v0.10.3 (C:\Users\me\Documents\nokhwa)
   Compiling nokhwactl v0.10.0 (C:\Users\me\Documents\nokhwa\examples\capture)
    Finished dev [unoptimized + debuginfo] target(s) in 10.57s
     Running `C:\Users\me\Documents\nokhwa\target\debug\nokhwactl.exe list-properties 0 All`
Nokhwa Initalized: true
Controls for camera 0
Control: Brightness, Name: Brightness, Value: (Current: 128, Default: 128, Step: 1, Range: (0, 255)), Flag: [Manual], Active: true
Control: Contrast, Name: Contrast, Value: (Current: 32, Default: 32, Step: 1, Range: (0, 255)), Flag: [Manual], Active: true
Control: Hue, Name: Hue, Value: (Current: 0, Default: 0, Step: 1, Range: (-180, 180)), Flag: [Manual], Active: true
Control: Saturation, Name: Saturation, Value: (Current: 64, Default: 64, Step: 1, Range: (0, 100)), Flag: [Manual], Active: true
Control: Sharpness, Name: Sharpness, Value: (Current: 2, Default: 2, Step: 1, Range: (0, 7)), Flag: [Manual], Active: true
Control: Gamma, Name: Gamma, Value: (Current: 120, Default: 120, Step: 1, Range: (90, 150)), Flag: [Manual], Active: true
Control: WhiteBalance, Name: WhiteBalance, Value: (Current: 4000, Default: 4000, Step: 1, Range: (2800, 6500)), Flag: [Automatic], Active: true
Control: BacklightComp, Name: BacklightComp, Value: (Current: true, Default: true), Flag: [Manual], Active: true
Control: Pan, Name: Pan, Value: (Current: 0, Default: 0, Step: 1, Range: (-16, 16)), Flag: [Manual], Active: true
Control: Tilt, Name: Tilt, Value: (Current: 0, Default: 0, Step: 1, Range: (-16, 16)), Flag: [Manual], Active: true
Control: Zoom, Name: Zoom, Value: (Current: 100, Default: 100, Step: 10, Range: (100, 400)), Flag: [Manual], Active: true
Control: Exposure, Name: Exposure, Value: (Current: -6, Default: -6, Step: 1), Flag: [Automatic], Active: true
MJPEG:
 - 320x180: [30, 30, 30, 30]
 - 320x240: [30, 30, 30, 30]
 - 352x288: [30, 30, 30, 30]
 - 424x240: [30, 30, 30, 30]
 - 640x360: [30, 30, 30, 30]
 - 640x480: [30, 30, 30, 30]
 - 848x480: [30, 30, 30, 30]
 - 960x540: [30, 30, 30, 30]
 - 1280x720: [30, 30, 30, 30]
YUYV:
 - 320x180: [30, 30, 30, 30]
 - 320x240: [30, 30, 30, 30]
 - 352x288: [30, 30, 30, 30]
 - 424x240: [30, 30, 30, 30]
 - 640x360: [30, 30, 30, 30]
 - 640x480: [30, 30, 30, 30]
 - 848x480: [20, 20, 20, 20]
 - 960x540: [15, 15, 15, 15]
 - 1280x720: [10, 10, 10, 10]
NV12:
 - 320x180: [30, 30, 30, 30]
 - 320x240: [30, 30, 30, 30]
 - 352x288: [30, 30, 30, 30]
 - 424x240: [30, 30, 30, 30]
 - 640x360: [30, 30, 30, 30]
 - 640x480: [30, 30, 30, 30]
 - 848x480: [30, 30, 30, 30]
 - 960x540: [30, 30, 30, 30]
 - 1280x720: [30, 30, 30, 30]
GRAY:
RAWRGB:
 - 320x180: [30, 30, 30, 30]
 - 320x240: [30, 30, 30, 30]
 - 352x288: [30, 30, 30, 30]
 - 424x240: [30, 30, 30, 30]
 - 640x360: [30, 30, 30, 30]
 - 640x480: [30, 30, 30, 30]
 - 848x480: [20, 20, 20, 20]
 - 960x540: [15, 15, 15, 15]
 - 1280x720: [10, 10, 10, 10]
```

But here's what it returns when I create camera with arbitrary format:

```rust
let requested_format = RequestedFormat::new::<RgbFormat>(RequestedFormatType::None);
let camera = nokhwa::Camera::new(camera_info.index().clone(), requested_format)?;
println!("{:?}", camera.camera_format());
```

```
CameraFormat { resolution: Resolution { width_x: 1280, height_y: 720 }, format: NV12, frame_rate: 1 }
```

Doesn't matter which `RequestedFormatType` I choose, result is the same. If I retrieve `frame_rate()` directly, result is also the same, always `1`.

k-wasniowski, 2023-06-08:
I have encountered the same issue.

k-wasniowski, 2023-06-16:
I could resolve it. Could you add me as a maintainer?
@l1npengtul

l1npengtul, 2023-06-16:
Sure, I need maintainers especially for different platforms as I use linux exclusively.

JosephCatrambone, 2023-09-06:
@k-wasniowski Can you confirm/deny that it's the same root cause as #139 ?  I.e., the let fps in msmf_backend is trying to grab an oddly converted FPS?

I closed that ticket as a duplicate, but if it's not I'll reopen it.

NigamLad, 2026-10-08:
I recently had this issue on my system and recorded many videos at 1 fps. Just wondering if anyone know any techniques to recover frames using some interpolation techniques, or something else to generate missing frames