l1npengtul/nokhwa #126: Compile issue on linux

DomtronVox, 2023-06-14:
### OS
Linux, Ubuntu 22.04.2 LTS

### Rust Version
```
$ rustup update
info: syncing channel updates for 'stable-x86_64-unknown-linux-gnu'
info: syncing channel updates for 'nightly-x86_64-unknown-linux-gnu'
info: checking for self-update

   stable-x86_64-unknown-linux-gnu unchanged - rustc 1.70.0 (90c541806 2023-05-31)
  nightly-x86_64-unknown-linux-gnu unchanged - rustc 1.72.0-nightly (371994e0d 2023-06-13)

info: cleaning up downloads & tmp directories
```

### Cargo.toml Dependency line
`nokhwa = {version = "0.10.4", features = ["input-native", "output-wgpu"]}`

### Compile Error
``` rust
   Compiling nokhwa-core v0.1.1
error[E0308]: mismatched types
   --> /home/domtron/.cargo/registry/src/index.crates.io-6f17d22bba15001f/nokhwa-core-0.1.1/src/traits.rs:236:32
    |
236 |                 bytes_per_row: width_nonzero,
    |                                ^^^^^^^^^^^^^ expected `Option<u32>`, found `Option<NonZeroU32>`
    |
    = note: expected enum `Option<u32>`
               found enum `Option<NonZeroU32>`

error[E0308]: mismatched types
   --> /home/domtron/.cargo/registry/src/index.crates.io-6f17d22bba15001f/nokhwa-core-0.1.1/src/traits.rs:237:33
    |
237 |                 rows_per_image: height_nonzero,
    |                                 ^^^^^^^^^^^^^^ expected `Option<u32>`, found `Option<NonZeroU32>`
    |
    = note: expected enum `Option<u32>`
               found enum `Option<NonZeroU32>`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `nokhwa-core` (lib) due to 2 previous errors
```

### Reproduction
* $ cargo new test_nokhwa
* add `nokhwa = {version = "0.10.4", features = ["input-native", "output-wgpu"]}` to Cargo.toml dependencies section. 
* $ cargo run

### Comments
I tried all of the different versions between 0.10.0 to 0.10.4 and 0.9.2. Also tried several specific commits to see if I could find one that worked. 

I am trying to build this with eframe to use along side egui, but the error shows up in a fresh project with just nokhwa as indicated in the Reproduction section. 

Really confused on what I'm doing wrong.

OlivierLDff, 2023-06-20:
Just use github dependency:
```toml
[dependencies.nokhwa]
git = "https://github.com/l1npengtul/nokhwa.git"
rev = "ea971242a888987222375189fb78c84fa85e5a4e"
```

stolinski, 2023-06-20:
> Just use github dependency:
> 
> ```toml
> [dependencies.nokhwa]
> git = "https://github.com/l1npengtul/nokhwa.git"
> rev = "ea971242a888987222375189fb78c84fa85e5a4e"
> ```

Dang. Still getting the same with that version.

```
Compiling nokhwa-core v0.1.2 (https://github.com/l1npengtul/nokhwa.git?rev=ea971242a888987222375189fb78c84fa85e5a4e#ea971242)
error[E0308]: mismatched types
   --> /Users/scotttolinski/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/ea97124/nokhwa-core/src/traits.rs:230:32
    |
230 |                 bytes_per_row: width_nonzero,
    |                                ^^^^^^^^^^^^^ expected enum `Option`, found `u32`
    |
    = note: expected enum `Option<u32>`
               found type `u32`
help: try wrapping the expression in `Some`
    |
230 |                 bytes_per_row: Some(width_nonzero),
    |                                +++++             +

error[E0308]: mismatched types
   --> /Users/scotttolinski/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/ea97124/nokhwa-core/src/traits.rs:231:33
    |
231 |                 rows_per_image: height_nonzero,
    |                                 ^^^^^^^^^^^^^^ expected enum `Option`, found `u32`
    |
    = note: expected enum `Option<u32>`
               found type `u32`
help: try wrapping the expression in `Some`
    |
231 |                 rows_per_image: Some(height_nonzero),
    |                                 +++++              +

For more information about this error, try `rustc --explain E0308`.
error: could not compile `nokhwa-core` due to 2 previous errors
```

DomtronVox, 2023-06-20:
> @DomtronVox did you find a work around? Also stuck here.

I did not. Even tried going into the code to fix the specific error but it just made more errors. But I haven't messed with it since then.

However, it does _look_ like OlivierLDff's suggestion fixed it for me, maybe.

> Just use github dependency:
> 
> ```toml
> [dependencies.nokhwa]
> git = "https://github.com/l1npengtul/nokhwa.git"
> rev = "ea971242a888987222375189fb78c84fa85e5a4e"
> ```

I don't have time to test it more right now, but it seems to have built fine using that specific checkout.

stolinski, 2023-06-20:
It builds for me until I add `features = ["input-native", "output-wgpu"]` then it fails with the same error.

OlivierLDff, 2023-06-20:
I haven't tried wgpu feature so yes maybe it is broken

anselanza, 2023-06-22:
Wow, I'm really struggling to get any kind of example working. I'm also on MacOS.

After lots of fiddling I settled on 
```
[dependencies]
nokhwa = { version = "=0.10.3", features=["input-native","output-threaded"]}
```
which at least got rid of compile-time errors.

Then I tried to use the example from `./examples/threaded-capture/src/main.rs` and that compiled but immediately returns runtime errors such as `thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ProcessFrameError { src: NV12, destination: "RGB", error: "bad input buffer size" }', src/main.rs:47:56`

mischa-robots, 2024-06-24:
Jetson Nano, Arm64, Ubuntu18 (L4T)

I can confirm that this does not compile with the above error:

```
[dependencies.nokhwa]
version = "0.10.4"
features = ["input-native", "output-wgpu"]
```

While this does compile:

```
[dependencies.nokhwa]
version = "0.10.4"
features=["input-native","output-threaded"]
```

@anselanza thank you a lot for the hint.

airstrike, 2025-05-08:
Also struggling to get examples working. `cargo run --example` or `cargo run --package` should work.

commit 7176265c (2023-06-15): update nokhwa core fix #126