l1npengtul/nokhwa #130: Compile issue in capture example

PuffyWithEyes, 2023-06-19:
Compile issue in ```capture``` example on Raspberry Pi 4. Please fix it for simple start

- error[E0432]: unresolved import `nokhwa::CallbackCamera`
-   --> src/main.rs:96:13
-    |
- 96 |     Buffer, CallbackCamera, Camera,
-    |             ^^^^^^^^^^^^^^ no `CallbackCamera` in the root
- 
- error: cannot find derive macro `Parser` in this scope
-    --> src/main.rs:135:10
-     |
- 135 | #[derive(Parser)]
-     |          ^^^^^^
-     |
- note: `Parser` is imported here, but it is only a trait, without a derive macro
-    --> src/main.rs:78:12
-     |
- 78  | use clap::{Parser, Subcommand};
-     |            ^^^^^^
- 
- error: cannot find attribute `command` in this scope
-    --> src/main.rs:136:3
-     |
- 136 | #[command(author, version, about, long_about = None)]
-     |   ^^^^^^^
- 
- error: cannot find attribute `command` in this scope
-    --> src/main.rs:138:7
-     |
- 138 |     #[command(subcommand)]
-     |       ^^^^^^^
- 
- error: cannot find derive macro `Subcommand` in this scope
-    --> src/main.rs:159:10
-     |
- 159 | #[derive(Subcommand)]
-     |          ^^^^^^^^^^
-     |
- note: `Subcommand` is imported here, but it is only a trait, without a derive macro
-    --> src/main.rs:78:20
-     |
- 78  | use clap::{Parser, Subcommand};
-     |                    ^^^^^^^^^^
- 
- error[E0599]: no function or associated item named `parse` found for struct `Cli` in the current scope
-    --> src/main.rs:309:20
-     |
- 137 | struct Cli {
-     | ---------- function or associated item `parse` not found for this struct
- ...
- 309 |     let cli = Cli::parse();
-     |                    ^^^^^ function or associated item not found in `Cli`
-     |
-     = help: items from traits can only be used if the trait is implemented and in scope
-     = note: the following traits define an item `parse`, perhaps you need to implement one of them:
-             candidate #1: `Parser`
-             candidate #2: `TypedValueParser`
-             candidate #3: `object::read::elf::file::FileHeader`
-             candidate #4: `object::read::macho::file::MachHeader`
-             candidate #5: `object::read::pe::file::ImageNtHeaders`
-             candidate #6: `nom::internal::Parser`
-             candidate #7: `nom::sequence::Tuple`
-             candidate #8: `ttf_parser::ggg::layout_table::RecordListItem`
-             candidate #9: `ttf_parser::ggg::lookup::LookupSubtable`
-             candidate #10: `ttf_parser::parser::FromData`
-             candidate #11: `ttf_parser::parser::FromSlice`
- 
- error[E0614]: type `PropertyKind` cannot be dereferenced
-    --> src/main.rs:324:28
-     |
- 324 |                 Some(k) => *k,
-     |                            ^^
- 
- Some errors have detailed explanations: E0432, E0599, E0614.
- For more information about an error, try `rustc --explain E0432`.
- error: could not compile `cam_rs` (bin "cam_rs") due to 7 previous errors

OlivierLDff, 2023-06-20:
Have you tried 0.10 branch ?

PuffyWithEyes, 2023-06-23:
I tried compile all branches. But every example in branch can't compile

closed 2025-03-10 by PuffyWithEyes