l1npengtul/nokhwa #134: Not compiling on mac os.

The-Real-Thisas, 2023-07-21:
## Cargo.toml

```toml
[package]
name = "test"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]

[dependencies.nokhwa]
version = "0.10.0"
# Use the native input backends, enable WGPU integration
features = ["input-native", "output-wgpu"]
```

## main.rs 

```rust
fn main () {
    // first camera in system
    let index = CameraIndex::Index(0); 
    // request the absolute highest resolution CameraFormat that can be decoded to RGB.
    let requested = RequestedFormat::new::<RgbFormat>(RequestedFormatType::AbsoluteHighestFrameRate);
    // make the camera
    let mut camera = Camera::new(index, requested).unwrap();

    // get a frame
    let frame = camera.frame().unwrap();
    println!("Captured Single Frame of {}", frame.buffer().len());
    // decode into an ImageBuffer
    let decoded = frame.decode_image::<RgbFormat>().unwrap();
    println!("Decoded Frame of {}", decoded.len());
}
```

## Output

```log
(base)  ~/Desktop/cs/rust/ [master] cargo run
    Blocking waiting for file lock on package cache
    Blocking waiting for file lock on package cache
    Blocking waiting for file lock on package cache
   Compiling nokhwa-core v0.1.2
   Compiling wgpu-hal v0.14.1
error[E0308]: mismatched types
   --> /Users/thisas/.cargo/registry/src/github.com-1ecc6299db9ec823/nokhwa-core-0.1.2/src/traits.rs:230:32
    |
230 |                 bytes_per_row: width_nonzero,
    |                                ^^^^^^^^^^^^^ expected enum `Option`, found `u32`
    |
    = note: expected enum `Option<u32>`
               found type `u32`
help: try wrapping the expression in `Some`
    |
230 |                 bytes_per_row: Some(width_nonzero),
    |                                +++++             +

error[E0308]: mismatched types
   --> /Users/thisas/.cargo/registry/src/github.com-1ecc6299db9ec823/nokhwa-core-0.1.2/src/traits.rs:231:33
    |
231 |                 rows_per_image: height_nonzero,
    |                                 ^^^^^^^^^^^^^^ expected enum `Option`, found `u32`
    |
    = note: expected enum `Option<u32>`
               found type `u32`
help: try wrapping the expression in `Some`
    |
231 |                 rows_per_image: Some(height_nonzero),
    |                                 +++++              +

For more information about this error, try `rustc --explain E0308`.
error: could not compile `nokhwa-core` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
(base)  ~/Desktop/cs/rust/ [master] 
```

pcontezini, 2023-07-26:
Same thing here, to fix (not tested thought) just do what the compiler says:
Line 230: bytes_per_row: Some(width_nonzero),
Line 231: rows_per_image: Some(height_nonzero),

Kofituo, 2023-10-13:
Facing this issue as well

ahirner, 2023-10-14:
I think this can be closed after https://github.com/l1npengtul/nokhwa/pull/142 and thus is duplicated by #141.

mangoleaf, 2023-11-25:
I agree that it can be closed, but it would also be nice if 0.10.5 crate was released so people don't have to pull directly from the git repo for the fix

Kofituo, 2023-11-27:
@mangoleaf @ahirner Actually, this issue still stands even to the date. I'm using M1 mac if that helps.

mangoleaf, 2023-12-12:
@Kofituo The fix was merged in #142, but the rust crate was never released since. This worked for me when I checked out the latest code instead of trying to get the crate via crates.io. I am also on an M1 mac.

Kofituo, 2023-12-12:
@mangoleaf @ahirner That's surprising. Here's my full project https://github.com/Kofituo/sample_gui/blob/master

Here's the full list of errors I'm facing. They're mostly syntax errors, among others:

```
// cargo run
   Compiling nokhwa-core v0.2.0 (https://github.com/l1npengtul/nokhwa.git#f0bbbf32)
error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> /Users/kofiotuo/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/f0bbbf3/nokhwa-core/src/format_request.rs:20:35
    |
20  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> /Users/kofiotuo/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/f0bbbf3/nokhwa-core/src/format_request.rs:45:35
    |
45  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> /Users/kofiotuo/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/f0bbbf3/nokhwa-core/src/format_request.rs:60:35
    |
60  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> /Users/kofiotuo/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/f0bbbf3/nokhwa-core/src/format_request.rs:76:35
    |
76  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> /Users/kofiotuo/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/f0bbbf3/nokhwa-core/src/format_request.rs:91:35
    |
91  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> /Users/kofiotuo/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/f0bbbf3/nokhwa-core/src/format_request.rs:115:35
    |
115 |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
[... 245 lines cut ...]
    |
126 |                 Self
    |                 ^^^^ help: a local variable with a similar name exists (notice the capitalization): `self`
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0223]: ambiguous associated type
  --> /Users/kofiotuo/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/f0bbbf3/nokhwa-core/src/decoder.rs:14:37
   |
14 |     type Container: Deref<Target = [Pixel::Subpixel]>;
   |                                     ^^^^^^^^^^^^^^^
   |
help: if there were a type named `Example` that implemented `Pixel`, you could use the fully-qualified path
   |
14 |     type Container: Deref<Target = [<Example as Pixel>::Subpixel]>;
   |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Some errors have detailed explanations: E0223, E0411, E0412, E0425, E0432, E0433.
For more information about an error, try `rustc --explain E0223`.
error: could not compile `nokhwa-core` (lib) due to 30 previous errors

```

MohammadKurjieh, 2024-02-26:
@Kofituo I was facing the same issue on mac m1, but using the pull #142, commit "8987af8", It compiles fine on my machine.