l1npengtul/nokhwa #135: Cannot compile on Windows and Linux at current HEAD ( 4c75bd7e2f77dc40dcfb9f8baa67475955d111a1 )

naari3, 2023-07-22:
I created this issue because no other issue was similar.
```
nokhwa on  senpai [!] is 📦 v0.11.0 via 🦀 v1.71.0 took 3s
❯ cargo check
    Checking nokhwa-core v0.2.0 (C:\Users\naari\src\github.com\naari3\nokhwa\nokhwa-core)
error[E0433]: failed to resolve: use of undeclared crate or module `serde`
 --> nokhwa-core\src\decoder.rs:3:5
  |
3 | use serde::de::Error;
  |     ^^^^^ use of undeclared crate or module `serde`

warning: unused import: `image::ImageBuffer`
  --> nokhwa-core\src\buffer.rs:19:5
   |
19 | use image::ImageBuffer;
   |     ^^^^^^^^^^^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: unused import: `crate::error::NokhwaError`
  --> nokhwa-core\src\buffer.rs:20:5
   |
20 | use crate::error::NokhwaError;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0223]: ambiguous associated type
  --> nokhwa-core\src\decoder.rs:14:37
   |
14 |     type Container: Deref<Target = [Pixel::Subpixel]>;
   |                                     ^^^^^^^^^^^^^^^
   |
help: if there were a type named `Example` that implemented `Pixel`, you could use the fully-qualified path
   |
14 |     type Container: Deref<Target = [<Example as Pixel>::Subpixel]>;
   |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Some errors have detailed explanations: E0223, E0433.
For more information about an error, try `rustc --explain E0223`.
warning: `nokhwa-core` (lib) generated 2 warnings
error: could not compile `nokhwa-core` (lib) due to 2 previous errors; 2 warnings emitted
```
I tested on windows-msvc and linux (Ubuntu on WSL2)
```
❯ rustc --version --verbose
rustc 1.71.0 (8ede3aae2 2023-07-12)
binary: rustc
commit-hash: 8ede3aae28fe6e4d52b38157d7bfe0d3bceef225
commit-date: 2023-07-12
host: x86_64-pc-windows-msvc
release: 1.71.0
LLVM version: 16.0.5
```
```
$ rustc --version --verbose 
rustc 1.71.0 (8ede3aae2 2023-07-12)
binary: rustc
commit-hash: 8ede3aae28fe6e4d52b38157d7bfe0d3bceef225
commit-date: 2023-07-12
host: x86_64-unknown-linux-gnu
release: 1.71.0
LLVM version: 16.0.5
```

naari3, 2023-07-22:
And the most recent commit I was able to compile was 557eba6c04e5137613e9b142102f029553d2dac2

gadirom, 2023-07-28:
Same here. Has anybody managed to overcome this?

JosephCatrambone, 2023-09-03:
Not so far.  I've been starting at the most recent and moving back, but I might take a break.  Like others have noted, serde missing sometimes.  mod utils missing in a few places.  Haven't found one after 0.10.3 other than 557eba6.