l1npengtul/nokhwa #143: Error compiling example code.

mostafamshkfides, 2023-10-01:
Hey guys,
I'm trying to build and run a simple Cargo project that uses this crate to capture a frame with my laptop camera.
I'm using ubuntu 20.04 LTS as my OS and this is my code in `main.rs` :
```
use nokhwa::utils::CameraIndex;
use nokhwa::utils::RequestedFormat;
use nokhwa::pixel_format::RgbFormat;
use nokhwa::utils::RequestedFormatType;
use nokhwa::Camera;

fn main() {

    // first camera in system
    let index = CameraIndex::Index(0); 
    // request the absolute highest resolution CameraFormat that can be decoded to RGB.
    let requested = RequestedFormat::new::<RgbFormat>(RequestedFormatType::AbsoluteHighestFrameRate);
    // make the camera
    let mut camera = Camera::new(index, requested).unwrap();

    // get a frame
    let frame = camera.frame().unwrap();
    println!("Captured Single Frame of {}", frame.buffer().len());
    // decode into an ImageBuffer
    let decoded = frame.decode_image::<RgbFormat>().unwrap();
    println!("Decoded Frame of {}", decoded.len());
}

```

But when I run `cargo run --release` , I got this error : 
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: NotImplementedError("Platform requirements not satisfied. (No Selection)")', src/main.rs:16:52
```

and I can't figure it out. 
I would appreciate it if you can help me.

RakuJa, 2023-10-02:
Did you specify in the Cargo.toml the input? 
For example

`
nokhwa = {version = "0.10.4", features = ["input-native"]}
`

G0rocks, 2024-02-20:
I didn't and was getting the error "This operation is not implemented yet: Platform requirements not satisfied. (No Selection)" when running:
`    // make the camera
    let mut camera = match Camera::new(camera_index, requested){
        Ok(t) => {t},
        Err(e) => {println!("Camera creation error:\n{}", e);
                                return;},
    };`

But adding the input in the cargo.toml file helped, added the answer to grepper, thank you!