l1npengtul/nokhwa #147: Compile issues on v4l2_pix_format in linux kernel 6.2

chriamue, 2023-10-13:
On kernel `Linux 6.2.0-34-generic` or maybe already earlier, v4l2 introduced unions, which does not work with `v4l2-sys-mit = "0.2"`:

```rust
/*
 *      V I D E O   I M A G E   F O R M A T
 */
struct v4l2_pix_format {
        __u32                   width;
        __u32                   height;
        __u32                   pixelformat;
        __u32                   field;          /* enum v4l2_field */
        __u32                   bytesperline;   /* for padding, zero if unused */
        __u32                   sizeimage;
        __u32                   colorspace;     /* enum v4l2_colorspace */
        __u32                   priv;           /* private data, depends on pixelformat */
        __u32                   flags;          /* format flags (V4L2_PIX_FMT_FLAG_*) */
        union {
                /* enum v4l2_ycbcr_encoding */
                __u32                   ycbcr_enc;
                /* enum v4l2_hsv_encoding */
                __u32                   hsv_enc;
        };
        __u32                   quantization;   /* enum v4l2_quantization */
        __u32                   xfer_func;      /* enum v4l2_xfer_func */
};
```

So I get this compile error like this:

```bash
Compiling v4l2-sys-mit v0.2.0
error: failed to run custom build command for `v4l2-sys-mit v0.2.0`

Caused by:
  process didn't exit successfully: `/home/chriamue/temp/test_v4l/target/debug/build/v4l2-sys-mit-ca66454ad2257fe1/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at /home/chriamue/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.69/src/fallback.rs:817:9:
  "v4l2_pix_format_union_(anonymous_at_/usr/include/linux/videodev2_h_500_2)" is not a valid Ident
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Upgrading to
```cargo
[target.'cfg(target_os="linux")'.dependencies]
v4l = "0.14"
v4l2-sys-mit = "0.3"
```
allows to compile.

Please upgrade in `nokhwa-bindings-linux` if possible.

ripytide, 2024-02-08:
Fixed by the as yet unmerged PR #156

faern, 2024-02-25:
What's the status on this issue? It's still broken for me on Fedora 39.

Skgland, 2024-02-26:
I think this might be fixed since f1405412e5e45f04339670d2f04aa88ccadb486b 8 month ago as that contains the dependency upgrade from #156 mentioned by ripytide in https://github.com/l1npengtul/nokhwa/issues/147#issuecomment-1933994342 , but the last release is 10 month old. So I think we might be just waiting for a new release of the 0.10 branch.

closed 2024-12-03 by l1npengtul