l1npengtul/nokhwa #148: Can't compile on archlinux 6.5.5-arch1-1

WyeneCloud, 2023-10-13:
Hi thanks you for taking time to read this issues.
first of all i'm on arch linux with the kernel 6.5.5-arch1-1. my cargo is cargo 1.72.0 (103a7ff2e 2023-08-15). i'm on hyprland.
my computer is an Lenovo Legion 5. 

here what i did :
``` 
( ~ )  ››› git clone https://github.com/l1npengtul/nokhwa.git
Cloning into 'nokhwa'...
remote: Enumerating objects: 3541, done.
remote: Counting objects: 100% (993/993), done.
remote: Compressing objects: 100% (345/345), done.
remote: Total 3541 (delta 640), reused 938 (delta 611), pack-reused 2548
Receiving objects: 100% (3541/3541), 7.13 MiB | 376.00 KiB/s, done.
Resolving deltas: 100% (2232/2232), done.
  ( ~ )  ››› cd nokhwa/
  ( nokhwa )  ››› cd examples/
  ( examples )  ››› cd capture/
  ( capture )  ››› ls
Cargo.toml  Info.plist  Makefile  README.md  example-capture  src
  ( capture )  ››› cargo run --release
```

and errors i got 
```
error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> nokhwa-core/src/format_request.rs:20:35
    |
20  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> nokhwa-core/src/format_request.rs:45:35
    |
45  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> nokhwa-core/src/format_request.rs:60:35
    |
60  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> nokhwa-core/src/format_request.rs:76:35
    |
76  |                         self.$name: Option<Range<$range_type>> = Some(Range {
    |                                   ^ expected one of 7 possible tokens
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error: expected one of `(`, `.`, `::`, `;`, `?`, `}`, or an operator, found `:`
   --> nokhwa-core/src/format_request.rs:91:35
[... 238 lines cut ...]

error[E0411]: cannot find type `Self` in this scope
   --> nokhwa-core/src/format_request.rs:124:93
    |
124 |             pub fn [< with_ $name _range >](mut self, $name: Option<Range<$range_type>>) -> Self {
    |                    ------------------------ `Self` not allowed in a function                ^^^^ `Self` is only available in impls, traits, and type definitions
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0425]: cannot find value `Self` in this scope
   --> nokhwa-core/src/format_request.rs:126:17
    |
126 |                 Self
    |                 ^^^^ help: a local variable with a similar name exists (notice the capitalization): `self`
...
210 | range_set_fields!((Resolution, resolution), (u32, frame_rate),);
    | --------------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `range_set_fields` (in Nightly builds, run with -Z macro-backtrace for more info)
```
(i hope i make an good issues with all information needed)
i really dont know what i did wrong. thanks you

Kofituo, 2023-10-16:
I also have this issue with the latest branch on Mac. Downgrading is probably the best option (ie build the `0.10` branch instead)

l1npengtul, 2024-12-03:
0.11 doesnt compile yet
0.10 should work for you.
Open new issue if problem persists, thank you!