l1npengtul/nokhwa #161: Meta: Current Status, Progress and Future Plans for this Project

marcpabst, 2024-01-22:
Hey @l1npengtul!

First off, a big shoutout to you and all the contributors for your incredible work on nokhwa. It's such a valuable tool for webcam/camera capture and honestly, it really stands out from the alternatives that are currently available in rust like the bindings to gstreamer and openpnp capture.

I've noticed that the last release on crates.io dates back over a year, and there are a handful of unresolved issues in the repo. I'm really keen on contributing and helping nokhwa evolve, but at the moment, it seems a bit challenging. I personally think the main hurdles are: a) it's not quite clear if PRs are being reviewed actively, and b) the split between two parallel branches makes the project's direction a bit ambiguous.

I'd love to get involved and support nokhwa, but I'd like to understand the current state of the project first. Could you share whether the project is still a focus and, if so, what the upcoming plans are? This info would be incredibly valuable for me and possibly other potential contributors.

Thanks again for all your work on this project!

(also pinging some other people who might be interested in this discussion - please feel free to weigh in and share your thoughts: @yamt, @michivi, @hanguk0726, @alufers, @olonho, @vyeevani)

l1npengtul, 2024-01-22:
Currently, I am in college so it is a bit different for me to work on it at the moment. I get on break in feburary and plan to resume work on 0.11

The most daunting task in my opinion is the MacOS backend. I don't know what I was doing when I made it, and I don't want to touch it but obviously work must be done.

l1npengtul, 2024-01-22:
My plans for 0.11 are outlined in #86.

l1npengtul, 2024-01-22:
0.10 is a dead branch, with only some band-aids applied while I work on 0.11.

marcpabst, 2024-01-23:
Thank you for your reply! It's good to hear that you're still interested in maintaining this project. Believe me, I know how much work and responsibility these kinds of projects can be!

I think my main problem currently is that I can't get the senpai branch to compile (I'm on MacOS), so I'm continuing to use 0.10 while I would love to fix the MacOS backend. Which is currently completely broken on my MacBook - neither the internal webcam works (I think it's outputting some weird 10bit NV12 format) nor external USB webcams do.

l1npengtul, 2024-01-23:
Thank you for your interest - I will try to get senpai to a compiling state ASAP.

alufers, 2024-01-23:
> nor external USB webcams do.

To get external webcams to work on macOS you need this fix: https://github.com/l1npengtul/nokhwa/pull/157

```toml
nokhwa = { git = "https://github.com/alufers/nokhwa.git", branch = "fix/0.10.3-fix-macos-usb-camera" }
```
This branch contains the fix, + fix for RGB conversion.

marcpabst, 2024-01-23:
> Thank you for your interest - I will try to get senpai to a compiling state ASAP.

That would be great! From there, I'll probably be able to get the MacOS backend in a working state!

ericreg, 2024-03-16:
@marcpabst  @l1npengtul  I am also very interested in contributing to this project, specifically for the linux / mac portions. I'm somewhat new to rust, but I have a few years experience in writing camera drivers.

@marcpabst  Do you have a branch for any changes you have made to the macOS backend?

marcpabst, 2024-03-17:
Hi, unfortunately, there hasn't been any progress as far as I can tell. And I've been quite busy with other things so no work on my side either. Still think this would make for a very useful package if in a usuable state...

Anyway, things should have become much easier on recent MacOS versions, in that MacOS does not expose any raw formats anymore (as far as I can tell) but converts them into a few standard pixel fromats regardless of what the camera provides. So no B/W support or raw MJPEG support through native MacOS APIs.

The-Real-Thisas, 2024-04-07:
@l1npengtul I can try my best to get it working on macos specifically on m1.

The-Real-Thisas, 2024-04-07:
But is the crate being updated ? Because #134 seemly fixed in source but not in crate.io

eduramiba, 2024-11-28:
I also want to thank you for this great library 😃
I recently used nokhwa to build a native library with good performance for webcam access in Java: https://github.com/eduramiba/webcam-capture-driver-native

If I can help somehow to progress on 0.11 please let me know