l1npengtul/nokhwa #162: Support for macOS continuity camera

gianzellweger, 2024-02-03:
I tried to test this crate on macOS on a mac without a camera, so I connected my iPhone using [Continuity Camera](https://support.apple.com/en-us/102546) to see what it could do. Running the capture example with the `list-devices` option however resulted in this:
```
2024-02-03 22:54:30.163 nokhwactl[1428:2290738] WARNING: AVCaptureDeviceTypeExternal is deprecated for Continuity Cameras. Please use AVCaptureDeviceTypeContinuityCamera and add NSCameraUseContinuityCameraDeviceType to your Info.plist.
There are 0 available cameras.
```
I would probably classify this as `P3`. It would be nice to have, but doesn't have a high priority as it is a pretty niche use-case. I would love to help, but am probably unable to do so, due to my lack of understanding of pure-Apple development (Objective-C, Swift, NS-Anything, etc).

Danss1mo, 2024-02-11:
Translate it, I so lazy to do it myself))))) But it will solve your problem

Кажется, что вы получаете предупреждение относительно использования устаревшего AVCaptureDeviceTypeExternal в своем коде на PyCharm.  

Чтобы исправить это, вам нужно обновить AVCaptureDeviceTypeExternal на AVCaptureDeviceTypeContinuityCamera и добавить NSCameraUseContinuityCameraDeviceType в ваш Info.plist файл.

Вот как вы можете это сделать:

1. Откройте свой проект в PyCharm.
2. Найдите файл Info.plist в проекте. Обычно он находится в папке с названием вашего проекта.
3. Откройте файл Info.plist в текстовом редакторе.
4. Найдите ключ NSCameraUsageDescription (если его нет, то создайте его).
5. Добавьте значение NSCameraUseContinuityCameraDeviceType к ключу NSCameraUsageDescription.
6. Сохраните изменения в файле Info.plist.
7. Перезапустите PyCharm и попробуйте запустить ваш код еще раз. 

Это должно привести к устранению предупреждения. Если предупреждение продолжает появляться, убедитесь, что вы правильно изменили Info.plist файл и перезапустили PyCharm.

Надеюсь, это поможет вам решить вашу проблему! Если у вас есть еще вопросы, не стесняйтесь задавать.

gianzellweger, 2024-02-11:
I don't know the PList format, could you expand on what you mean by "Добавьте значение NSCameraUseContinuityCameraDeviceType к ключу NSCameraUsageDescription"? Could you show a concrete code example?

marcpabst, 2024-02-14:
This definitely used to work. I think a recent MacOS update broke this by changing the APIs for internal/external camera access. Should be relatively straightforward to fix this.

This will probbaly be addressed when the MacOS portion of this crate gets rewritten, but the maintainer is currently busy and whle I would love to help, I'm also swamped and quite happy with my gstreamer solution for now. One day tho...

DocDanM, 2025-01-09:
Using Mac OS Sequoia 15.2 and running this command in terminal **ffmpeg -f avfoundation -list_devices true -i ""** gives this output **WARNING: Add NSCameraUseContinuityCameraDeviceType to your Info.plist to use AVCaptureDeviceTypeContinuityCamera** I don't know what to do now. I don't know where to find this info.plist file. I don't know if it exists. I don't know how to edit it. (I think it has to do with Apples intention to block unauthorized access to the cameras on devices, which sounds like a good thing.)

l1npengtul, 2025-01-09:
Info.plist is part of the application and not this library, you will likely need to build and package with xcode to get it to work.