l1npengtul/nokhwa #163: Possible unsoundness

manglemix, 2024-02-21:
I am running my program which uses `nokhwa` with valgrind to solve an unrelated issue, but this warning popped up:
```
==179483== Syscall param ioctl(VKI_V4L2_S_FMT) points to uninitialised byte(s)
==179483==    at 0x551A88D: syscall (syscall.S:38)
==179483==    by 0xA119E6: v4l::v4l2::api::detail::ioctl (api.rs:80)
==179483==    by 0xA14476: v4l::v4l2::api::ioctl (api.rs:206)
==179483==    by 0xA06318: v4l::video::capture::<impl v4l::video::traits::Capture for v4l::device::Device>::set_format (macros.rs:158)
==179483==    by 0x9D9AE6: nokhwa_bindings_linux::internal::V4LCaptureDevice::new (lib.rs:238)
==179483==    by 0x9D2FC4: nokhwa::camera::init_v4l (camera.rs:461)
==179483==    by 0x9D23D6: nokhwa::camera::init_camera (camera.rs:571)
==179483==    by 0x9D1B39: nokhwa::camera::Camera::with_backend (camera.rs:55)
==179483==    by 0x9D1ACD: nokhwa::camera::Camera::new (camera.rs:44)
==179483==    by 0x9CADB3: camera::Camera::new (lib.rs:33)
==179483==    by 0x75B930: core::iter::traits::iterator::Iterator::count (lib.rs:92)
==179483==    by 0x760EE5: _ZN7lunabot4main28_$u7b$$u7b$closure$u7d$$u7d$17h381408d88255a9caE.llvm.14441783956877976536 (main.rs:41)
==179483==  Address 0x1ffeff50f4 is on thread 1's stack
==179483==  in frame #3, created by v4l::video::capture::<impl v4l::video::traits::Capture for v4l::device::Device>::set_format (macros.rs:152)
```
I am unsure how to interpret this.

RivenSkaye, 2024-03-20:
Last line in the output seems to suggest this is coming from the pointer returned by v4l? Could try running some other app using it (ffmpeg will probably work) to see if it still happens

l1npengtul, 2024-10-12:
This seems like a V4L2 issue, not a nokhwa issue.

closed 2024-10-12 by l1npengtul