l1npengtul/nokhwa #164: Switching devices leads to assertion failure at the device list.

caseykneale, 2024-03-03:
I tried to write some code where I switched devices with the V4L backend. IE: 
- I capture an image on device index 0. Effectively following an index.
- Close the stream.
- Create a new camera (say index 2)
- open a stream
- Capture a new image

When I get to the open stream phase of this  I get an assertion failure:
```
thread '<unnamed>' panicked at /home/cake/.cargo/git/checkouts/nokhwa-66b597f0abb1b795/8ac5ce9/nokhwa-bindings-linux/src/lib.rs:193:13:
assertion `left == right` failed: Device list should not contain duplicate indexes
```

Ordinarily I would share code but this is easily reproduced. I've tried pretty much every pattern to avoid this. From mutating the camera instance by changing it's device id (same error), not closing the streams, not opening a new stream etc. 

Ultimately I'm not sure why a global device list is creating a new device entry. I tried inspecting the docs for what happens when a stream is closed, it says there are quirks and to check the backend portion of the docs. Those notes seem to be missing?

Is there a preferred method to switch devices, is this currently unsupported, or broken?

caseykneale, 2024-03-03:
For anyone wondering the OpenCV crate does support this functionality without any issues.

wangjia184, 2024-09-08:
I am encountering the same issue on Ubuntu

```
# https://github.com/l1npengtul/nokhwa/issues/177
nokhwa = { version = "0.10.4", features = ["input-native"], git = "https://github.com/kendfrey/nokhwa.git", branch = "0.10" }
```

wangjia184, 2024-09-08:
The bug is from these lines: https://github.com/wangjia184/nokhwa/blob/8890670bf107fe7252d416a480dbe696de4211a8/nokhwa-bindings-linux/src/lib.rs#L200-L207

A fix is here : https://github.com/kendfrey/nokhwa/compare/0.10...wangjia184:nokhwa:0.10

caseykneale, 2024-09-08:
Thanks for the fix and the reply I would say this is now closed :).

closed 2024-09-08 by caseykneale

robryk, 2025-04-25:
The most recent version of nokhwa-bindings-linux published on crates.io still has this issue, as well as the tip of the 0.10 branch in this repository.