l1npengtul/nokhwa #174: Failed to find tool. Is `aarch64-linux-gnu-gcc` installed?

zxcqirara, 2024-06-03:
Just created a new project, added this lib and tried to compile it. It throws this error:
```
The following warnings were emitted during compilation:

warning: mozjpeg-sys@1.1.1: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `aarch64-linux-gnu-gcc` installed?
warning: mozjpeg-sys@1.1.1: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `aarch64-linux-gnu-gcc` installed?

error: failed to run custom build command for `mozjpeg-sys v1.1.1`

Caused by:
  process didn't exit successfully: `/home/user/proj/target/debug/build/mozjpeg-sys-ef7e4aece9a75277/build-script-build` (exit status: 1)
  --- stdout
  cargo:include=/home/user/proj/target/aarch64-unknown-linux-gnu/debug/build/mozjpeg-sys-dca8f2dbe12942ef/out/include:/home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/mozjpeg-sys-1.1.1/vendor
  cargo:lib_version=62
  TARGET = Some("aarch64-unknown-linux-gnu")
  OPT_LEVEL = Some("0")
  HOST = Some("x86_64-unknown-linux-gnu")
  cargo:rerun-if-env-changed=CC_aarch64-unknown-linux-gnu
  CC_aarch64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CC_aarch64_unknown_linux_gnu
  CC_aarch64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=TARGET_CC
  TARGET_CC = None
  cargo:rerun-if-env-changed=CC
  CC = None
  RUSTC_LINKER = None
  cargo:rerun-if-env-changed=CROSS_COMPILE
  CROSS_COMPILE = None
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:warning=Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `aarch64-linux-gnu-gcc` installed?
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("true")
  CARGO_CFG_TARGET_FEATURE = Some("neon")
  cargo:rerun-if-env-changed=CFLAGS_aarch64-unknown-linux-gnu
  CFLAGS_aarch64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CFLAGS_aarch64_unknown_linux_gnu
  CFLAGS_aarch64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=TARGET_CFLAGS
  TARGET_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:warning=Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `aarch64-linux-gnu-gcc` installed?
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  CARGO_CFG_TARGET_FEATURE = Some("neon")
  cargo:rerun-if-env-changed=CFLAGS_aarch64-unknown-linux-gnu
  CFLAGS_aarch64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CFLAGS_aarch64_unknown_linux_gnu
  CFLAGS_aarch64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=TARGET_CFLAGS
  TARGET_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None

  --- stderr

  error occurred: Failed to find tool. Is `aarch64-linux-gnu-gcc` installed?

warning: build failed, waiting for other jobs to finish...
error: failed to run custom build command for `v4l2-sys-mit v0.2.0`

Caused by:
  process didn't exit successfully: `/home/user/proj/target/debug/build/v4l2-sys-mit-844fdf7983ccc2a0/build-script-build` (exit status: 101)
  --- stderr
  /usr/include/gnu/stubs.h:7:11: fatal error: 'gnu/stubs-32.h' file not found
  /usr/include/gnu/stubs.h:7:11: fatal error: 'gnu/stubs-32.h' file not found, err: true
  thread 'main' panicked at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/v4l2-sys-mit-0.2.0/build.rs:10:10:
  Failed to generate bindings: ()
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I'm on x64 linux and trying to compile it for aarch64 linux, and I **have** installer `aarch64-linux-gnu-gcc`

zxcqirara, 2024-06-03:
Tried to cross build but it still throws an error:
```
error: failed to run custom build command for `mozjpeg-sys v1.1.1`

Caused by:
  process didn't exit successfully: `/target/debug/build/mozjpeg-sys-ef7e4aece9a75277/build-script-build` (exit status: 1)
  --- stderr
  /target/debug/build/mozjpeg-sys-ef7e4aece9a75277/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /target/debug/build/mozjpeg-sys-ef7e4aece9a75277/build-script-build)
  /target/debug/build/mozjpeg-sys-ef7e4aece9a75277/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /target/debug/build/mozjpeg-sys-ef7e4aece9a75277/build-script-build)
  /target/debug/build/mozjpeg-sys-ef7e4aece9a75277/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /target/debug/build/mozjpeg-sys-ef7e4aece9a75277/build-script-build)
warning: build failed, waiting for other jobs to finish...
```

unwx, 2024-07-03:
I encountered the same build failure. The root cause explained here: https://stackoverflow.com/a/76443281

I'm new to Rust, but it looks like:

- The v4l crate updated its bindgen dependency from 0.59 to 0.65 to solve that problem: https://github.com/raymanfx/libv4l-rs/commit/cf6d4125d3f37f24d18c325fd369f503d9f12a88
- nokhwa updated its v4l dependency on June 4th, 2023: https://github.com/l1npengtul/nokhwa/commit/f1405412e5e45f04339670d2f04aa88ccadb486b
- However, the latest nokhwa package on [crates.io](https://crates.io/crates/nokhwa) was updated on May 14th, 2023, and doesn't include this fix.

---

**Workaround**:

Building nokhwa from source (the [0.10](https://github.com/l1npengtul/nokhwa/tree/0.10) branch) resolved the issue for me.

**Additional Notes**:

On Fedora, I needed to install the following development packages for the build to succeed: 

`fontconfig-devel, alsa-lib-devel, libudev-devel`

l1npengtul, 2024-10-12:
0.10.5 has updated the libv4l-rs dependency. Should be fixed, please feel free to open a new issue if issue persists

closed 2024-10-12 by l1npengtul